c0pperdragon/Amiga-Digital-Video #7: Support for other systems

IanSB, 2020-09-17:
It might be possible to implement support for other relatively simple systems, maybe with a buffer board and just Pi software or possibly using the 14 bits available on CPLD with the new 12 bit board in a similar way to the proposed direct 48K Spectrum connection.

A few machines that this might work with:
Atari 2600 
VIC 20
Commodore 16 / +4 (TED chip)

c0pperdragon, 2020-09-17:
I did a quick research on the matter, and the machines you mentioned all use a graphics chip that directly generates the usual set of television signals (luma and chroma) that are basically impossible to convert back to a pixel-perfect digital representation (I tried with the Atari800XL and the C64 and failed spectacularly).

So the only option is of course to use the digital inputs. In the case of the ZX Spectrum there are not so many relevant signals and it is pretty straight-forward how to map this to the video image. For the aforementioned 3 machines this is not so easy. All of those have a rich set of internal registers which influence the video output in many weird and wonderful ways. I guess none of those is as complicated as the VIC-II, but still it requires some substantial computation if done entirely in software and at a tight time-budget.
More complicated is the fact that these chips have many data and address inputs as well, so 12 (or 14) signals lines are never enough to capture all that is relevant. 

Maybe the Atari 2600 could be done by using the technique of an existing RGB mod: Intercept the writes to the color registers and replace them with spoofed colors. After that whenever the TIA writes the 3-bit lum value of one of the registers to the digial lum lines, you can know exactly which register the value came from and you know with which "real" color to replace it.
This of course disturbs the analog signal generation. You also need some fast CPLD that is able to drive the TIAs input lines.
This is the original RGB mod: http://etim.net.au/2600rgb/

IanSB, 2020-09-17:
> All of those have a rich set of internal registers which influence the video output in many weird and wonderful ways

I guess the only way for those to work then would be a similar FPGA solution like the C64 / 800XL although at least the VHDL would be a bit simpler.

> Maybe the Atari 2600 could be done by using the technique of an existing RGB mod: Intercept the writes to the color registers

That could be an interesting solution as it might be just a relatively simple board with a CPLD that plugs into the TIA socket with a Pi plugged into that. Not sure about clearances though

c0pperdragon, 2020-09-18:
For an overall solution to mod the Atari 2600 for HDMI, you really need to include audio as well into the HDMI output. The TIA produces only a digital beeping. This could fairly easily be sampled by the same CPLD and transfered to the Pi. No idea what software-wizardry is then needed to get usable sound to HDMI.

IanSB, 2020-09-18:
>For an overall solution to mod the Atari 2600 for HDMI, you really need to include audio as well into the HDMI output

I've not really looked into the audio side of things, I guess the same thing applies to the C64, Amiga and Atari computers as well. 

Most of the original computers supported by the scaler had built in speakers so no requirement for audio and in my case I tend to use it with old 1600x1200 4:3 monitors with DVI input and an analog soundbar so again no benefit from HDMI audio.

You can DMA ingest PCM audio through GPIO's 18-21 using one of the Pi's built in peripherals but those GPIO pins are being used for something else at the moment so it would require a complete redesign and a larger form factor to add the audio chips.

Most HDMI monitors have a PC mode where you can connect analog audio but the other workaround would be to use a HDMI audio inserter although they seem to be expensive compared to audio extractors.

c0pperdragon, 2020-09-19:
In my personal setup I currently also have analog audio only. For external converter solutions this may still be the best option, otherwise you would have an additional conversion step from analog to digital and back again.
For internal solutions where you can directly tap into a digital data stream this is a different matter.
While the analog sound synthesis of the C64 and probably the Atari ST cN not be digitally captured, the Atari 2600, Atari 800 (?) and, most important of all, the Amiga have digital output from the sound chips.
Full-blown 100% digital HDMI from the Amiga would be even more awesome than what we already have. I also think off ease of video capturing here.

c0pperdragon, 2020-09-19:
No, I was mistaken. The Paula chip in the Amiga already has the DAC stage built in. So no digital audio stream there. 
Same thing is probably true for the Atari 800 (guessing from the schematics - did not probe the pins of the POKEY yet). 

So the only machines from where the audio can actually be taken in a pure digital form are the ZX Spectrum and the Atari 2600.  
Both have a sound quality so bad that many games do not even bother to play any.

Coincidentially I have recently modded a 2600 to composite video + audio and just added a simple low-pass filter stage to the audio. This now sounds not half as bad as before.  Still no winner...

culexus8, 2021-01-01:
Anyone know if this might work on consoles like Nes, snes, Sega?

c0pperdragon, 2021-01-01:
This specific solution (a Pi Zero with only a small bit of extra electronics) is only possible if the computer already provides the pixel data in digital form on some accessable leads. This is true for the Amiga and the Atari ST.  
But in your mentioned consoles the analog video signal is generated inside the graphics chips and it is normally impossible to convert this back to a glitch-free digital signal.  
There are quite some projects to do something about that but they are much more complex.

bandaidboi, 2021-01-11:
@c0pperdragon Regarding the use of the PiZero with your schematics for the Amiga Adapter, would your design be easily adaptable to the Atari ST/STE? i.e. all 12 color bits from the GSTSHIFTER, csync from else where on the board and the use of the 8mhz system clock?

c0pperdragon, 2021-01-12:
I just re-opened the according thread:  https://github.com/c0pperdragon/Amiga-Digital-Video/issues/6
There was substantial work done already by @IanSB for this machine.  I will try to give a summary of my thoughts in this thread.

closed 2021-02-28 by c0pperdragon