c0pperdragon/Amiga-Digital-Video #9: Amiga Interlaced mode is off by one line

c0pperdragon, 2020-09-19:
Hi @IanSB

As previously mentioned: When using 1280x1024 output resolution, an interlaced workbench is off by one interlaced line.

[image]

This can also not be fixed by adjusting the vertical offset, because then the picture is too up.
Even if it could be fixed by adjustements, I guess, the non-interlaced picture would be off instead.

For larger output resolutions this effect is basically the same, but there it is not so noticable:

[image]
[image]

IanSB, 2020-09-20:
Can you try this version:

[kernelrpi.zip](https://github.com/c0pperdragon/Amiga-Digital-Video/files/5252708/kernelrpi.zip)

c0pperdragon, 2020-09-21:
The Interlace mode looks like it has a balanced top and bottom overscan now with 1680x1050 output resolution. I can not switch to 1280x1024, because the single-button mode does no longer work after swapping out the kernel file. Do I have to make any changes to my setup to re-activate this feature?

IanSB, 2020-09-21:
The way single button mode is selected was changed: Please update to the current release version first then replace the kernel file with the above:
https://github.com/hoglet67/RGBtoHDMI/releases
https://github.com/hoglet67/RGBtoHDMI/releases/download/20200916_6310e8c/RGBtoHDMI_20200916_6310e8c.zip

c0pperdragon, 2020-09-21:
Interlaced display is perfect now.
[image]

I have adjusted my configuration to  the "Trailing -ve" sampling option in preparation for the upcomming SuperDenise-adjusted adapter board. If I got everything correct, this configuration should directly work  with it as well. I have also reduced the overscan a bit because the right area was a bit narrower where the Amiga set the last two pixels to black already.
Also when running my Agnus in NTSC mode, the right overscan ends with a strange jagged border. So 712 pixels is now my maximum width that should fit all purposes.
[Amiga.txt](https://github.com/c0pperdragon/Amiga-Digital-Video/files/5256852/Amiga.txt)

IanSB, 2020-09-21:
>I have adjusted my configuration to the "Trailing -ve" sampling option

Yes, the above kernel has the corrected text for the sampling option.

BTW the menus are a very flickery in interlaced mode because they are only written on alternate fields.

closed 2020-09-23 by c0pperdragon