c0pperdragon/Amiga-Digital-Video #11: User Feedback and Experiences

c0pperdragon, 2020-11-19:
I created this thread as a means to collect feedback about your experiences using this solution. Installation pictures and screenshots are highly welcome.

Markcalibrav6, 2020-11-20:
[image]

 Tested on a A500+ with a TF536(TF536/Lazarus relocator)  OS3.1 4 Rom workbench 3.9 CGX(native driver) Hi Res laced 16 colours.Standard A500 psu

The adapter was well made and compact design....the jumper on the back is perfectly placed to enable selection of OCS/ECS Chipset....😁

The RPI Zero takes its power from the Amiga so no psu is needed on the RPI side....😉

Easy SD card setup...makes this adapter a dream to setup and use...first power on and you are thrown into a Awesome pixel perfect screen...crisp clear colours.

WHDload games all work well so far...with no artifacts or ghosting at all...no lag no issues when screen modes are changed....however only standard OCS/ECS screen modes are currently  supported....no wide screen options yet or Super72/SuperPlus(800x600)

My Indivison ECS V2 has config tools to tweek the settings that are saved to the device....this would be nice on the RPI side.

Mabe a short GPIO extension cable would be good as the RPI Zero faces the 68000 socket....add the HDMI lead and there is no room for popular CPU cards for the A500...(TF series..with relocator) however if your CPU card can fit directly into the 68000 socket you might be onto a winner.

Overall this adapter is a great option given its low cost compared to other options out there.

c0pperdragon, 2020-11-20:
You can optionally connect a push button to the two angled pins on the adapter board. With such a button you can call up a settings menu (long press) and switch through the various options and modify quite some things (using short presses and long presses). This rich feature set comes from the RGBtoHDMI software that can do much more than just support the Amiga.
But for the Amiga I found that there is normally not much use in trying to tweak the settings. Output resolution is already autodetected and the 4x upscaling looks 100% crisp and accurate. But you may want to turn on scan-line overlay or some interpolation filter. So you actually can do this. Settings are stored on the SD card.

Markcalibrav6, 2020-12-01:
Added a button to the Zero....super cool picture....

https://youtu.be/-gxpG8moEEI

ryanm101, 2021-01-14:
With wicher508i HDMI cable is a tight fit, I've ordered some right angle adaptors to see if that helps.

dahabakuk, 2021-01-14:
i wonder if anyone is using an turbocard with this mod... looks like the hdmi cable will be a really tight fight? can someone share a picture with turbocards installed?

Osmund, 2021-01-16:
Tried to get it to fit with a tf card but there is no clearance. Is it possible to rotate the header for the pi zero 180 degrees, so that the hdmi points to the back of the amiga?

c0pperdragon, 2021-01-17:
I don't have a design for this. I original thought about stacking the Pi directly on top of the Denise, but this would become a problem for the heat dissipation, so I now used this more spread out layout. 
But I have designed a smaller adapter which can be used more universall,y but needs soldering of many individual wires.  If you can do that, this may be the solution for you  (but I don't build these boards either).

gaspafen, 2021-01-20:
With such a waiting list for ordering one I ordered some PCB's from eurocircuits and parts from mouser. Had a local supply of the RPi. Forgot to order the resistors so I scrounged a 5k from old broken electronics. Put it all together and it works great. Fits fine in the A500 case together with the vampire v2+. This could be because the pin headers I used have a bit longer pins. Will be ordering some other pin headers to use if I solder another together. Need to get a HDMI switch now :)
[image]
[image]

Osmund, 2021-01-20:
I was lucky and secured a couple before the Jan Beta videos. To make it fit with my tf534 I soldered the pinheader on the pi zero on the bottom with one row on the outside of the adapter. I then connected the outside pins to the second row on the pi. I can now mount the pi to the adapter with the hdmi facing towards the back of the case. With a tower of pisa stack of cpu sockets, the tf534 fits over the pi. It works great, have not had any termal issues. The amiga top case fits as well.
[image]
[image]
[image]

de-nugan, 2021-01-28:
I've built two V2 boards from PCBWay. Mine have a couple of differences to the original design:
1)  I couldn't source the exact regulators so used these => https://au.rs-online.com/web/p/low-dropout-voltage-regulators/7968322/
2) The Pi is connected with an extremely short (~1cm) IDC cable for now so it fits with the Kipper 2K 8MB+IDE board in the A500 and also because I can't find the 40-pin female IDC PCB headers anywhere around here :(

Both boards show a sharp and clean image at the 3.1 boot screen, but both have sparkly interference patterns on a grey workbench screen. The colour "sparkles" are mostly green and red, and appear on the trailing edge of black to grey (left to right) transitions. I've seen one or two references to minor cases of this but mine is pretty severe and impossible to ignore. 

A few tests so far:
- Change length of the IDC cable to the Pi => sparkles seem to get worse with a longer cable, lines start to flicker more
- Change from original to ATX PSU => no change in symptoms
- Connect additional ground from Pi to A500 chassis => no change

So far I can't tell if the problem is some colour bits staying low for an additional clock cycle OR additional bits going high. I'm wondering if any of the engineers here could offer some insights as the the cause of this problem? 

A couple of additional issues are the top scan line flickering left to right, and the geometry is off, but I'm assuming these can be tuned out. 

Really impressed by this project though, kudos to everyone involved!

[image]

[image]

hansliss, 2021-01-28:
I ordered PCBA cards from JLCPCB and they look really good. I used 74LVC574APW, 74LVC86APW,118 and TLV74333PDBVR from LCSC - I had to solder on the voltage regulator myself since the PCBA service didn't have those in stock. It all looks very professional, including my own solder work. :)

As far as I could determine, the silkscreen markings on JP1 are wrong; for a Denise (such as my A2000 has), JP1 should be strapped as 2-3, judging from the schematic. If I don't do that, I get unsynchronized garbage - the correct colours, but it scans very slowly and appears to be lacking a clock.

With JP1 strapped 2-3, I get a picture, but it's flashing slowly on and off, has a few artifacts, and isn't properly positioned. Any ideas? I managed to get a screenshot.

[image]

/Hans

c0pperdragon, 2021-01-28:
@de-nugan :  There are two general sources of pixel errors:  
1. The latch in the adapter is triggered too close to the time when color lines change. This can be adjusted by moving the Denise/SupeDenise jump to the other position.
2. The color lines going into the Pi are not stable enough when the clock input to the Pi toggles. This can be caused by serveral things:  
2a. Higher capacitance/inductance caused by the longer data path due to the ribbon cable. You could try to compensate that by artifically adding a small capacitor to the PiCLK line.
2b. Bad grounding. In your setup the GND for the Pi is proved through the ribbon cable. You could add some extra GND connection with a short (and reasonably thick wire) to some GND point on the main board.
2c. Different behaviour of the logic ICs. I only designed an tested the circuitry with parts from TI. Maybe parts from other manufacturers have different dynamic characteristics.

c0pperdragon, 2021-01-28:
@hansliss
I doublechecked JP1.  The setting 1-2 is indeed correct for a standard Denise. With the setting 2-3 it would use the inverted 7Mhz as clock which is the best choice for a SuperDenise.
 When the 1-2 setting does not work at all, there is something wrong with your adapter. Maybe solder bridging or no contact.

hansliss, 2021-01-28:
@c0pperdragon Thanks for checking! I'm just still not sure about JP1. Pin 1 is connected to Denise pin 34, right? The thing is, from the schematics for the A1000, on which my ancient German A2000 is apparently based, doesn't have pin 34 connected at all. It is indeed connected to CDAC on later models. Maybe that's the problem?

IanSB, 2021-01-28:
@hansliss 

A flashing screen usually means the sync timing doesn't match any of the available profiles.
What software version are you using?

The latest beta is here in case that helps:
https://github.com/IanSB/RGBtoHDMI/releases

hansliss, 2021-01-29:
> @hansliss
> 
> A flashing screen usually means the sync timing doesn't match any of the available profiles.
> What software version are you using?
> 
> The latest beta is here in case that helps:
> https://github.com/IanSB/RGBtoHDMI/releases

Thanks, I've tried the latest beta. No difference.

hansliss, 2021-01-29:
> @c0pperdragon Thanks for checking! I think the CDAC signal might be the problem, after all. The thing is, judging from the schematics for the A1000, on which my ancient German A2000 is apparently based, the Denise socket doesn't have pin 34 connected at all. It is indeed connected to CDAC on later models. Does that sound plausible?

I've checked the CDAC pin on the Zorro II bus now, and there's a ~7MHz clock signal there, as well as on pin 35 on Denise, but I didn't find anything on pin 34.

c0pperdragon, 2021-01-29:
Hmm, this could really mean that the very early model versions did not have the _CDAC signal connected to the Denise socket. 
I am not sure where you can find this signal on your specific machine, but you can probably figure this out with an oscilloscope and then retro-fit this signal to the Denise socket with a botch wire as well.

hansliss, 2021-01-29:
> Hmm, this could really mean that the very early model versions did not have the _CDAC signal connected to the Denise socket.
> I am not sure where you can find this signal on your specific machine, [...]

I'm going to start by verifying the adapter in an A500. I'm awaiting delivery now. I'll build a couple more adapters tomorrow and then I'll continue this once the A500 arrives. After that I may go back to the A2000 and try again - I don't like it right now. :)

hansliss, 2021-01-30:
> Hmm, this could really mean that the very early model versions did not have the _CDAC signal connected to the Denise socket.
> I am not sure where you can find this signal on your specific machine, [...]

On the Wikipedia page about A2000 (of all places), I found this: "[...] the early 2000 motherboard [...], cannot be upgraded with newer versions of the chipset [...]". To me, this confirms that what I've found may very well be correct - that there really is no CDAC signal on the Denise socket.

hansliss, 2021-01-30:
> Hmm, this could really mean that the very early model versions did not have the _CDAC signal connected to the Denise socket.
> I am not sure where you can find this signal on your specific machine, [...]

I just saw a schematic that was a bit legible than what I've seen before, and found that another signal is missing on the A1000/A2000-A motherboards: Pin 32 is NC instead of /CSYNC, which seems to be used by the RGBToHDMI adapter.

tk-amiga, 2021-01-31:
Made a few V1's and my own works very nicely in a rev6a motherboard with Super Denise. I guess V2 is more for the rev8 motherboard. Will get those soon too. 
https://nextcloud.amigaaa.com/s/q7dRdWoordnTmDY
https://nextcloud.amigaaa.com/s/23kHYkrD59GKCZm
Luckily my turboboard is in the sideslot.

IanSB, 2021-02-03:
@hansliss
>A1000/A2000-A motherboards: Pin 32 is NC instead of /CSYNC, which seems to be used by the RGBToHDMI adapter.
 
Yes, CSYNC is required, one thing that is a little confusing is that your screencap above is horizontally locked which it shouldn't be if sync was missing completely but it looks like the floating sync input is picking up crosstalk from the RGB signals and interpreting noise from the start of video as the edge of the sync pulse which would explain the shift to the left.

hansliss, 2021-02-03:
> I just saw a schematic that was a bit legible than what I've seen before, and found that another signal is missing on the A1000/A2000-A motherboards: Pin 32 is NC instead of /CSYNC, which seems to [...]

I got this working. See #28 for details.

IanSB, 2021-02-04:
@de-nugan 
>The colour "sparkles" are mostly green and red, and appear on the trailing edge of black to grey (left to right) transitions. 

What revision motherboard is your system?

de-nugan, 2021-02-05:
I think it’s rev 5 or 6, but can’t be sure as I’m away at the moment.

Also I found the flipflops I used were Nexperia parts. I have some TI parts on order to try when I get back mid-month.

lilwashu, 2021-02-18:
I have noticed "sparkling" but only on pink text in Indiana Jones and the Fate of Atlantis game. The "sparkles" are white. I'm running an A2000 rev 6.2 motherboard with a rev.1 adapter from seller "r3ne3gad3mast3r" on eBay and Terriblefire TF536 accelerator.

c0pperdragon, 2021-02-20:
Chances are that you have a SuperDenise on your board. This is not working well with the Rev.1 adapter. The Rev.2 has a jumper to select between original Denise and SuperDenise. 
If you are good with a soldering iron you could try to botch the adapter to 
make it somehow work. If you want I can give you instructions.

lilwashu, 2021-02-20:
Thanks for the reply - I have an 8362R8 OCS Denise. I haven't seen the issue before or since so I'll see if I can reproduce it, trying to do so in e.g. DPaint has so far failed. The installation is a pretty tight fit in the A2000 so I'll also see if removing the PSU frame makes any difference, there is light pressure on it.

FluffyRedLobster, 2021-02-27:
Reporting great success with installation alongside a TF536 accelerator on a rev 6a A500 😃 . This was with a v1 board but I'm sure the concept will apply to v2.

As others have found, space is tight if using a CPU relocator.  I considered options that others have mentioned in this thread - GPIO flex cable, or osmund's method of offsetting the headers and mounting the zero on it's back.  I experimented with using 2 x 90 degree pin headers to achieve something similar but was worried about the extra connections and additional signal distance given some of the reports here.

Ended up with a pretty simple solution - 2 sets of machine pin headers to lift the CPU relocator and the TF536 (one set for each gives the required clearance and still allows the keyboard to fit without fouling on the relocator), and _gently_ angle the GPIO pin headers on the adapter so the Pi Zero points toward the mainboard by 10 degrees or so.  The mini HDMI cable then outputs under the relocator to be routed to wherever it's needed.  I soldered the Pi directly to the angled pin headers but probably could have used a female header strip if I thought about it... it's not like the Pi is going anywhere else...  Lastly as I had a fairly chunky mini HDMI cable (the officialy Raspberry Pi one)I removed some of the plastic shroud to ensure there is no pressure on the solder joints from the cable.  I slimmer cable probably wouldn't need this.

The output quality is astonishing!  I'm using an old 17" Iiyama Prolite that syncs at 50Hz and scrolling is perfect too.  This easily beats the indivision mk2 in my A1200 in that regard.   Thanks c0pperdragon and hoglet67 for this great mod 👍 

Hope the approach helps some other CPU relocator users out there.

[image]

solarmon, 2021-03-07:
@c0pperdragon 

What did you mean by:

>    2a. Higher capacitance/inductance caused by the longer data path due to the ribbon cable. You could try to compensate that by artifically adding a small capacitor to the PiCLK line.

How would the cap be added? In parallel (with what?) or in series? What value would you propose?

IanSB, 2021-03-07:
For sparkly pixel related problems, especially with rev5 or earlier A500s or other models, try replacing the 74LVC parts with 74VHC parts. These are slower and fixed the sparkly problem on my rev5 which doesn't have the clock filtering components that are fitted on rev6 & later boards and which slightly delay the clock signals. You may get away with just changing U1 (74LVC86) to VHC so try that first.

Note they must be 74VHC parts which are 5v tolerant, not 74HC parts which are not 5v tolerant when powered at 3.3v

c0pperdragon, 2021-03-08:
@solarmon 
To make the clock signal to the Pi running a tiny bit slower, you could try to add a capacitor (lets say 10pF or a bit higher if this does not yet work) between pin 16 (GPIO23) and pin 0 (GND) of either the Pi or the adapter.

solarmon, 2021-03-08:
@c0pperdragon I only had "22" (27pF) ceramic cap to hand. This didn't seem to make any noticeable difference all (good or bad). So I'll have to source some smaller caps to try them out. When you sat 'or a bit higher' - how big a difference is this? Can I go from 10nF to 20nF, for example, or should the jump be smaller than that (i,e. 10, 12, 15, 18, 20, etc)?

c0pperdragon, 2021-03-08:
1 nF is for sure too much already. If 68pF does not yet work, I would give up on the idea.

solarmon, 2021-03-08:
Sorry, I should have said "10pF to 20pF". As stated, I've already tried 27pF.

c0pperdragon, 2021-03-08:
I did a bit of read-up on the matter of output capacitance for LVC parts, and it seems that to get a significant change in switching time you should actually use something like 200pF. This should delay the output for about 5ns which seems about right to me.

solarmon, 2021-03-08:
I had a bag of small 27pF ceramic caps. I put a load of them in parallel (probably about up to 30/40) and it didn't seem to make any noticeable difference at all. Not sure if what I'm doing is right - I thought putting caps in parallel adds up their capacitance.

c0pperdragon, 2021-03-08:
Yes, putting capacitors in parallel does add the capacitance. 
I am not sure what is going on here. I would expect it to make at least some difference, either for better or worse.
Could you post a picture or your installation?

de-nugan, 2021-03-10:
Following on from above, my Pi is now soldered to the adapter to get the shortest possible data path and this has successfully fixed the colour sparkles and glitches. The picture is now superb on my Rev 5 A500. 

I soldered the adapter board in place of the Denise socket for reliability and compactness, so it's now a permanent daughter board. A couple of caps and a resister pack needed to be nudged out of the way but overall it was a pretty easy fit and takes a minimum of space. The boards now sit low enough to go underneath the CPU expansion card without any additional risers, so the case can go back together as normal. 

I'm using a short mini male to std HDMI female cable from the Pi, it was necessary to cut the molding from the mini plug end to fit the cable between the ROM and CPU. It's tight but fits together pretty well. I have had some CF card errors since the installation, don't know yet if EMI from the adapter is causing issues with the expansion card above.. needs more testing. 

[image]
[image]
[image]

seb132, 2021-03-13:
Has Edu Arana (Arananet) released his PCB of the official repository ?
He said on Facebook that he would -- after he presented his very neat redesign of the adapter for the A500, compatible with Vampire, etc. But so far I haven't seen it yet, just in pictures.
[image]

solarmon, 2021-03-13:
@seb132 

As an alternative option, once I've tested my design, I will be releasing it.

[image]

dgoswick, 2021-03-18:
What does this mean?  I verified that my build is right.  This is on a stock Amiga 500.
[image]

mthiesen, 2021-03-19:
I can add another datapoint. I have ordered a board from eBay and installed it into my Amiga 500 Rev. 5. It works almost perfectly, but I am experiencing the same pixel sparkling that @lilwashu reported. It only seems to happen on very special color transitions and always from very dark to bright.

This is a screenshot from the Monkey Island intro:
[image]

I can also reproduce the issue with Deluxe Paint IV and the example image `ReferencePalette.LoRes`.  In this case only the medium saturation and low saturation rows are affected. The effect is by far the strongest in the last two boxes of the medium saturation row.
[image]

c0pperdragon, 2021-03-20:
When you have sparkling pixels, check out this thread for a very detailed discussion an possible solutions:
https://github.com/c0pperdragon/Amiga-Digital-Video/issues/40

jock-mcbufty, 2021-03-21:
Hi there!

When I have installed the GBItoHDMI and Raspberry Pi Zero on my A2000
[image]

i get this output! Any ideas anyone?

LinuxJedi, 2021-03-22:
> Hi there!
> 
> When I have installed the RGBtoHDMI and Raspberry Pi Zero on my A2000
> [image]
> 
> i get this output! Any ideas anyone?

Are you using the A2000 board or the original A500 board? Make sure you have the jumper set correctly for the Denise you have in the A2000 (some surprisingly do have a Super Denise in them when you least expect it). Beyond that your board may need some tweaks.

jock-mcbufty, 2021-03-22:
> Are you using the A2000 board or the original A500 board? Make sure you have the jumper set correctly for the Denise you have in the A2000 (some surprisingly do have a Super Denise in them when you [...]

—

Hi!

Yeah 2000 board Rev 6. I guess I can try popping the jumper back to Super Denise. I’m very much a super novice so if it’s soldering etc I will pass. The whole thing was propping up the PSU & FDD caddy anyway. Shame though it was a good idea. So this device is really only guaranteed for A500 & 1200s?

cheers!

hansliss, 2021-03-22:
> —
>
> Hi!
>
> Yeah 2000 board Rev 6. I guess I can try popping the jumper back to Super Denise. I’m very much a super novice so if it’s soldering etc I will pass. The whole thing was propping up the PSU & [...]

I'm awaiting 500mA regulators and will soon build five boards with those, for five different people (myself not included since my A2000 is to old for the video slot version). I will encourage all of them to write about their experiences here.

LinuxJedi, 2021-03-22:
> Hi!
>
> Yeah 2000 board Rev 6. I guess I can try popping the jumper back to Super Denise. I’m very much a super novice so if it’s soldering etc I will pass. The whole thing was propping up the PSU & [...]

If the Denise is a chip with the model 8373 (will have an R3 or R4 after it) then you have a Super Denise, if it is 8362 (likely an R8 after it) you have a regular Denise.

It should work with A2000, but there is a specific card version for it (which I'm working on an improved version of) that goes in the slot to the right of PSU as the front of the machine faces you instead of under it.

jock-mcbufty, 2021-03-22:
> If the Denise is a chip with the model 8373 (will have an R3 or R4 after it) then you have a Super Denise, if it is 8362 (likely an R8 after it) you have a regular Denise.
>
> It should work with [...]

—

I appreciate all the replies, I find modding fairly stressful when you are as clueless as myself. This setup looked fairly straight forward so I had a go. 

I will check tonight what chip I have, just in case! Again thanks for the tip regards versions. Can I quickly check that I haven’t fitted the whole thing the wrong way round? I made sure the slightly nick or circle on the side of the device matched the hole below it on the motherboard itself.

thanks!

jock-mcbufty, 2021-03-22:
[image]
It looks like I have a super Denise so I will switch jumpers and have another go!

hansliss, 2021-03-22:
> I will check tonight what chip I have, just in case! Again thanks for the tip regards versions. Can I quickly check that I haven’t fitted the whole thing the wrong way round? I made sure the slightly nick or circle on the side of the device matched the hole below it on the motherboard itself.
> 

Since you get a picture at all, I'm sure you've basically done it correctly. There are some weird issues with sparkling edges and similar that might or might not be related to what you get. For now, some suggested changes have been to use another technology for the three chips on the adapter board (74VHC* instead of 74LVC*) and (although this may be only for the video slot version) to use a beefier voltage regulator. I don't think there's any final solution for that particular problem, though.

hansliss, 2021-03-22:
> It looks like I have a super Denise so I will switch jumpers and have another go!

Oh, would be great if it's that simple.

LinuxJedi, 2021-03-22:
> Since you get a picture at all, I'm sure you've basically done it correctly. There are some weird issues with sparkling edges and similar that might or might not be related to what you get.

You would get a picture with the jumper set correctly, just a significant amount of noise as seen here.

jock-mcbufty, 2021-03-23:
[image]
[image]

hi guys, quick update - I swapped the jumper over to Super Denise to no avail. It still brings up a thin, distorted image. So I’ve gone back to my semi-reliable scart upscaler. For anyone who braves an A2000 mod, be aware that my PSU ‘tray’ touched the rasp pi. Thanks for all the replies guys!

hansliss, 2021-03-23:
> hi guys, quick update - I swapped the jumper over to Super Denise to no avail. It still brings up a thin, distorted image. So I’ve gone back to my semi-reliable scart upscaler. For anyone who braves an A2000 mod, be aware that my PSU ‘tray’ touched the rasp pi. Thanks for all the replies guys!

Before you give up, could you please do a quick test using a different monitor or TV? I've seen examples of when a monitor distorts the image - not quite that badly, but pretty bad.

lilwashu, 2021-03-23:
> [image]
> [image]
>
> hi guys, quick update - I swapped the jumper over to Super Denise to no avail. It still brings up a thin, distorted image. So I’ve gone back to my semi-reliable scart upscaler. For [...]

Apologies if this is stating the obvious but have you put insulating material (I used a thin bit of card and electrical tape) between the Pi pins and the metal PSU sled?

LinuxJedi, 2021-03-23:
> Apologies if this is stating the obvious but have you put insulating material (I used a thin bit of card and electrical tape) between the Pi pins and the metal PSU sled?

Beat me too it, those pins shorting on the metal there could definitely cause these kind of issues.

jock-mcbufty, 2021-03-23:
> > Apologies if this is stating the obvious but have you put insulating material (I used a thin bit of card and electrical tape) between the Pi pins and the metal PSU sled?
> 
> Beat me too it, those pins shorting on the metal there could definitely cause these kind of issues.

—

apologies that’s the output with the metal tray not on the pi. Yes if it worked I would have had to put something in between. Cheers for the answers though. It’s not just the distortion that’s the issue, it the thin resolution as well. Thanks.

[53 more replies]