c0pperdragon/Amiga-Digital-Video #12: Software update

IanSB, 2020-12-02:
Can you check out the latest version of RGBtoHDMI with the Amiga board to see that it still works OK. I've changed the menus so they don't flicker in interlaced mode by switching to progressive capture when the menus are on screen (The image under the menus will be at half vertical resolution).

The wiki has been updated to cover recent changes:
https://github.com/hoglet67/RGBtoHDMI/wiki/Quick-Start-Guide
https://github.com/hoglet67/RGBtoHDMI/wiki/Reference-Guide

Software:
https://github.com/hoglet67/RGBtoHDMI/releases

TuKo1982, 2020-12-02:
Works nicely for me

c0pperdragon, 2020-12-12:
I finally came around to put the Amiga on my bench again. The new firmware works flawlessly. I like the new option to test monitors for 50Hz compatibility. As you have announced, the interlace mode no longer flickers the settings menu (but shows only half the lines which is OK, I guess). So I updated the link to the firmware download.

Also I wanted to mention that the version 2 of the adapter seems to work fine with the SuperDenise, so I made a pull request for it to be included in the RGBtoHDMI repo.

closed 2020-12-25 by c0pperdragon