c0pperdragon/Amiga-Digital-Video #13: Adapter production

c0pperdragon, 2020-12-04:
I set up this thread to keep anyone informed who wants to buy an assembled adapter from me.
My first builds are nearly sold out (all ECS are gone or reserved) and I am in the process of ordering more parts. I will keep you informed.

culexus8, 2020-12-13:
I see you mention a V2 board in the readme. Is this for the A600?

c0pperdragon, 2020-12-14:
I sold the last part today, but expect more material to arrive in a few days.  
From now on I will only build the V2 variant which is OCS and ECS compatible.

c0pperdragon, 2020-12-25:
I spent parts of yesterday and today to assemble 10 Revison 2 adapters (feeling like a christmas elf all the while ;-) ) .

As it turns out, this is really labour-intensive so I need to adjust my prizing a bit (also to cover various overhead losses).
I can give out these adapters now for €29 plus shipping (€6 in Austria, €10 in Europe, €15 international, UK is currently not possible due to canceled air traffic).
Payment only via paypal to my account: reinhard.grafl@aon.at

Please reserve your adapter(s) by email and wait for my response before transfering any money.

If you cannot get any of these right now, do not worry. I will continue to make the adapters as long as there is demand.

Centrinouk, 2020-12-30:
Hi, interested in adapter but I'm from the UK, could you update when shipping is available please? Thanks

c0pperdragon, 2020-12-30:
My latest batch is already sold out. I expect more parts to arrive in a few days to make more adapters.

karynroberts, 2020-12-30:
> My latest batch is already sold out. I expect more parts to arrive in a few days to make more adapters.

Hi there - great project - I would like to get two of these :) thanks.

kipper2k, 2020-12-30:
Hi,  Is the Rev1 board same as rev2. i cannot find the gerbers for them,  are they public ?

 Thanks, very nice project

65Thumper, 2020-12-30:
Would like to say, great job! 
Also like to order 3 adapters, Thanks!!

c68696c66, 2020-12-30:
@kipper2k l found the v2 adapter gerbers here: https://github.com/c0pperdragon/RGBtoHDMI/tree/master/kicad_AmigaAdapter (looks like there is a pending pull request to merge it from this fork to the main repo...)

kipper2k, 2020-12-31:
found them. thanks for the info :)

nyteshade, 2020-12-31:
Would love a completed unit or two. Curious if there’s any known effort to get ECS/AGA support built?

amigacommodore, 2020-12-31:
I would like to order the adaptor but I live in Australia

c68696c66, 2020-12-31:
@kipper2k I opened the project in KiCad and generated the BOM. U4 seems to be a LDO Voltage Regulator TLV74333PDBVR

c0pperdragon, 2020-12-31:
Please do not clutter up this thread with off-topic stuff. Please create seperate threads for this.

warrengc3, 2021-01-01:
Hi, Any more ECS we can order?

65Thumper, 2021-01-03:
Yes, I still would like 3 for my Amiga 500s,  assemble of course, thanks!
E.Cotton

Gesendet mit meinem HTC

----- Nachricht beantworten -----

c0pperdragon, 2021-01-03:
Please send me an email so I can put you on my preorder list (which is growing quite long already).

amigacommodore, 2021-01-04:
I am yet to get a confirmation that you have put me on your list ?

c0pperdragon, 2021-01-05:
Please do not use this thread for reservations. I intend this striktly for announcments.  Send me an e-mail instead.

c0pperdragon, 2021-01-10:
I have finally received the PCBs I had ordered before christmas and was able to build 14 more adapters. With about 200 units on my reservation list already, this seems not much, but it is at least something. 
100 more PCBs are on their way and I am going to order more material  still..

I will contact the persons on the top of my list which are not in the UK (sorry, but there is still no mail service) and have sent send me an email before 31.12.2020.

c0pperdragon, 2021-01-27:
Have received 100 PCBs to build adapters from. With my current workflow I will be able to build about 16 parts in two days. I will send out notifications accordingly.

c0pperdragon, 2021-02-10:
Production is progresssing smoothly. So far I have processed all requests that were sent to me before 7.1. 20:00.
Hope to be able to catch up eventually....

c0pperdragon, 2021-02-26:
I have completed processing all queued user request and have a few pieces left. As I don't plan to build any more of these adapters these few can still be bought directly and after that you need to find other sources.

c0pperdragon, 2021-02-28:
During the last days I have sold off my remaining adapters. I don't plan to build any more of these in the forseeable future. As there are already other people taking up production, you may find such adapters on ebay or similiar platforms.

closed 2021-03-24 by c0pperdragon