c0pperdragon/Amiga-Digital-Video #15: BOM availability

kipper2k, 2020-12-31:
Hi, hopefully this post is in the correct place. Do you have a BOM publically available for this as i would like to build myself a couple of these. I seen the V1 pdf but not the V2 version.

 Thanks

culexus8, 2021-01-01:
I was also looking for the V2 and the small, looks like they are removed. It may be that he is doing some changes? I do have them at home but im at work for the moment.

zeakdad, 2021-01-10:
Loading the project into kicad, I get the below as a BOM via the eeschema tool but I've not gone through each item yet to see if there is enough detail to actually build up a proper items to order. Hopefully this will help though

Columns are Reference / Value / Footprint / Datasheet / Qty

v  C1 - C3	100nF	Capacitor_SMD:C_0603_1608Metric_Pad1.05x0.95mm_HandSolder	~	3
        C1	100nF	Capacitor_SMD:C_0603_1608Metric_Pad1.05x0.95mm_HandSolder	~	1
        C2	100nF	Capacitor_SMD:C_0603_1608Metric_Pad1.05x0.95mm_HandSolder	~	1
        C3	100nF	Capacitor_SMD:C_0603_1608Metric_Pad1.05x0.95mm_HandSolder	~	1
    C4	1uF	Capacitor_SMD:C_0805_2012Metric_Pad1.15x1.40mm_HandSolder	~	1
    JButton1	Conn_01x02	Connector_PinHeader_2.54mm:PinHeader_1x02_P2.54mm_Horizontal	~	1
>  JDenise1, JDenise2	Conn_02x12_Counter_Clockwise	Package_DIP:DIP-24_W15.24mm_Socket	~	2
>  JMainBoard1, JMainBoard2	Conn_01x24	Connector_PinHeader_2.54mm:PinHeader_1x24_P2.54mm_Vertical	~	2
    JP1	Jumper_3_Bridged12	Connector_PinHeader_2.54mm:PinHeader_1x03_P2.54mm_Horizontal	~	1
    JRaspberryPiZero1	Conn_02x20_Odd_Even	Connector_PinSocket_2.54mm:PinSocket_2x20_P2.54mm_Vertical	~	1
    R1	3k3	Resistor_SMD:R_0805_2012Metric_Pad1.15x1.40mm_HandSolder	~	1
    U1	74LVC86	Package_SO:TSSOP-14_4.4x5mm_P0.65mm	http://www.ti.com/lit/gpn/sn74HC86	1
v  U2, U3	74LVC574	Package_SO:TSSOP-20_4.4x6.5mm_P0.65mm	http://www.ti.com/lit/gpn/sn74HCT574	2
        U2	74LVC574	Package_SO:TSSOP-20_4.4x6.5mm_P0.65mm	http://www.ti.com/lit/gpn/sn74HCT574	1
        U3	74LVC574	Package_SO:TSSOP-20_4.4x6.5mm_P0.65mm	http://www.ti.com/lit/gpn/sn74HCT574	1
    U4	TLV74333PDBVR	Package_TO_SOT_SMD:SOT-23-5_HandSoldering	http://ww1.microchip.com/downloads/en/DeviceDoc/MIC550X.pdf	1

kipper2k, 2021-01-13:
Thanks for the info, it looks like i have enough info to build, i will leave this thread open till i can verify.

zeakdad, 2021-01-13:
I took a stab a selecting parts from digikey and ordered 10 boards from jlcpcb. I guess I'll see what happens when everything arrives. The passives I just had to guess the specs, everything else was pretty straight forward

kipper2k, 2021-01-13:
PCBs should be here on friday so i should hopefully be able to build. I have ordered the parts from digikey

Zucky16, 2021-01-15:
Kipper2k, if this works out for you please share your experience with us.  I remember a while back you were building a bunch of cool stuff for Amiga systems, but your funding dried up and you were forced to stop.  I'm in Ontario, Canada, if I remember correctly you weren't that far away.  If you are successful I'd definitely buy one of these boards from you, either fully assembled or just the bare board.

c0pperdragon, 2021-01-15:
Added a proper BOM to the repo

closed 2021-01-15 by c0pperdragon