c0pperdragon/Amiga-Digital-Video #17: BigBox Variants

nyteshade, 2021-01-06:
As I understand it, part of the problem for ECS and AGA machines is the processing power of the PiZero. I was wondering if @IanB might know if the processing power of the Pi CM4 or even a Pi4b attached to a card in a BigBox video slot might do the trick for the A2000s, A3000s and A4000 variants. The question came about after a discussion with Reinhard Grafl in which he mentioned

> OK, I get it. Looking at the schematics of the A2000 and A3000, the video slots provide all necessary signals from the (Super)Denise and power as well, so it would actually be possible to design an adapter 
> that fits in there and that can also carry the Raspberry Pi. Maybe someone actually wants to make such a board.
> For the A4000, the Lisa chip still can create much too much video data do be processed by the Pi. Restricting this to OCS screen modes would be a very unsatisfying solution.

Given this, and given the Pi CM4 (Compute Module 4) and Pi4B have more the double the resources of the PiZero, I was curious if this might be a future path. Also, given the CM4's size, it might be a more viable option for the A500s and A1000s of the world.

c0pperdragon, 2021-01-07:
@IanSB  is the person to ask

c0pperdragon, 2021-01-07:
As the compute modules do not have a video output port, this is not of much use here.
I am not sure about putting a Pi4B into an Amiga case.. The power consumption is pretty high, with corrsponding heat output. 
I don't know anything about the processing power and what it really takes to handle the ECS resolutions. 

To truly support AGA modes with 35ns pixel clock and 24 bit color depth the total data rate must be quadruppled compared to the current software. I can not imagine this to be possible in anything like the current approach.

closed 2021-01-15 by c0pperdragon