c0pperdragon/Amiga-Digital-Video #18: Can we take this external?

HS-Dave, 2021-01-07:
Sorry, don't know the best place to ask questions like this.

How dependent is your daughterboard on receiving the signals from denise pins directly? Could this potentially be made to work by taking output from the RGB out on the back and feeding that into your pcb to pass to the pi creating an external upscaling solution that should work with all denise chips/a1200/etc etc anything with an rgb out? Or is it that by the time it gets to the output port the signal is analog and no good for what you're doing?

IanSB, 2021-01-07:
The exernal version already works with the CPLD version of the board:
https://github.com/hoglet67/RGBtoHDMI/wiki/Bill-of-Materials-%2812-Bit-Board%29

You simply solder a 16 way ribbon cable to the denise chip and connect to the 12 bit extender board:
https://github.com/hoglet67/RGBtoHDMI/wiki/Bill-of-Materials-%2812-Bit-Extender%29

This version should also support some of the ECS modes because the CPLD can handle higher bandwidth 6 bits per pixel sources but I don't have an Amiga with an ECS chip so I haven't been able to test that so far.

HS-Dave, 2021-01-07:
thanks for the reply Ian :) What I meant by taking it external was remove the reliance on connections directly to internals like the denise chip and instead work from the RGB provided by the RGB-Video out port that is present on (I believe) basically all amigas regardless of chipset so you could easily swap between an A500 and an A1200 for example as you wished. Although as I was thinking about it I thought the problem here was that RGB-Out provides analog RGB where this mod is working specifically on the digital signals provided by denise, so maybe its a no goer. 

truthfully, I just want this mod for my AGA A1200 :)

c0pperdragon, 2021-01-07:
The analog outputs are always hard/impossible to convert back to their original digital counterparts for a pixel-perfect upscaling. Devices like the OSSC are trying to do this, but it takes quite some effort to get this working and the result may not be as perfect as a purely digital solution.

About AGA:
The pixel speed and color depth of the AGA resolutions are far beyond what can be done with this solution through the GPIO pins. An FPGA with a dedicated HDMI tranmitter chip may be the proper choice here. Of course this starts the be seriously expensive...

c0pperdragon, 2021-01-07:
@IanSB 
I don't know about the ECS modes. It is not just having 4 distinct colors plus the sprites that can appear anywhere with just any color. Also the palette can change totally unexpectedly anywhere.
So the Pi would need to be able to always handle the full 12 bit color depth at 35ns pixel clock. As the software is already struggling with a 70ns pixel clock, this is surely out of reach.

HS-Dave, 2021-01-08:
At this point we are basically building a DIY indivision AGA, I guess not for much cheaper (if cheaper at all). Ahhhh, poor 1200 :( 

> About AGA:
> The pixel speed and color depth of the AGA resolutions are far beyond what can be done with this solution through the GPIO pins. An FPGA with a dedicated HDMI tranmitter chip may be the proper choice here. Of course this starts the be seriously expensive...

closed 2021-01-15 by c0pperdragon