c0pperdragon/Amiga-Digital-Video #19: Issue with HDMI capture card

IanSB, 2021-01-07:
In his review, Jan Beta had a problem with his HDMI capture card.
The most likely cause of this is that the Pi zero is configured by default to output a DVI compatible RGB HDMI signal so that it can work with older 4:3 DVI monitors as well as more recent 16:9 HDMI monitors.

This can be changed by editing the config.txt and default_config.txt files.

Load both files into a text editor and change the following line from:
hdmi_drive=1
to 
hdmi_drive=2

Then save the files back to the SD card, reinsert into the Pi zero and reboot.
(Both files have to be edited because the config.txt file is occasionally re-created from the default_config.txt file when certain settings are changed)

If that doesn't work there are other hdmi settings that can be changed in the config file detailed here:
https://www.raspberrypi.org/documentation/configuration/config-txt/video.md

The one most likely to change things is hdmi_pixel_encoding:

0 | default (RGB limited for CEA, RGB full for DMT)
1 | RGB limited (16-235)
2 | RGB full (0-255)
3 | YCbCr limited (16-235)
4 | YCbCr full (0-255)

So try adding the following line just below the "hdmi_drive=" line
hdmi_pixel_encoding=3
or
hdmi_pixel_encoding=4

If someone with the problem can confirm that changing anything from the above fixes the issue  I'll add a menu option to switch between the various modes to save having to edit the text files.

c0pperdragon, 2021-01-07:
Contacted Jan to tell him of this possible solution...

JanBeta, 2021-01-08:
Hi, thanks so much for the help!

Changing the hdmi_drive=1 option to hdmi_drive=2 instantly fixed the problem.

For reference, I use an Elgato HD 60 S USB capture card and the colors were slightly off, with black displaying as green, when capturing video from the RaspberryPi output. No matter what settings I chose for the capturing card, the issue remained the same. Changing the above in the config.txt and default_config.txt on the Pi fixed it.

Again thanks for the help and of course for the amazing work from both of you. :)

Jan

IanSB, 2021-01-08:
> Changing the hdmi_drive=1 option to hdmi_drive=2 instantly fixed the problem.

Thanks for the confirmation
I've added a new "HDMI Mode" setting in the main menu with options:
DVI Compatible
HDMI (Auto RGB/YUV)
HDMI (RGB Limited)
HDMI (RGB Full)
HDMI (YUV Limited)
HDMI (YUV Full)

It will still default to DVI Compatible on first install as that ensures you get an image on all monitors.
If you change the setting to one of the HDMI options you will lose output on a DVI monitor. To recover from that you have to hold the button down during power up which will reset it back to DVI Compatible mode.

There will be a new release shortly with that and some other updates.
BTW to make it easier to update the software or retrieve screencaps (short presses of the switch screencap the image to a png file and write it to the SD card) you can use a micro SD extender and route that to the trapdoor on the A500 so you don't have to completely dismantle the Amiga.
e.g.
https://www.ebay.co.uk/itm/Micro-SD-To-TF-Card-Extension-Cable-Flexible-Extender-Adapter-Reader-Phone-UK-ti/383835410420
Check your local ebay for a suitable supplier.

IanSB, 2021-01-10:
@JanBeta 

There is a beta test version here with the above HDMI setting:
https://github.com/IanSB/RGBtoHDMI/releases/download/fdbf4a4/beta9.zip

This also has support for basic scanlines with the Amiga (They previously only worked at lower bit depths) although the level is not adjustable so they can only be black.
There is a setting in the preferences menu which now works with the Amiga:
"Swap Aspect 625<>525"
When set to "on" it changes the aspect ratio of the output to mimic the aspect ratio of the other standard which means that any Amiga games which were written for 525 Amigas can be displayed at full height 525 aspect ratio on a 625 line Amiga.
e.g:
Normal:
[image]
Swap Aspect set to "on":
[image]

Let me know if it works for you.

IanSB, 2021-01-10:
@JanBeta 

Updated to beta10 due to cache alignment timing issue:
https://github.com/IanSB/RGBtoHDMI/releases/download/626a527/beta10.zip

JanBeta, 2021-01-15:
Thanks for the updates! Tested the linked beta10 for some hours and everything seems to work as expected.

The 625/525 switching seemed to revert back sometimes (could only reproduce it once) when the software switched resolutions (I used Giana Sisters, title screen is interlaced, the rest is regular lo-res). Most of the times it worked without any problems.

The scanlines worked for me, too. Only tried it in low resolution though.

The slight flicker/sudden appearance of lines on the screen has completely disappeared (probably the timing fix you mentioned?), picture is rock stable and smooth now.

Switching HDMI modes works flawlessly. Video capture works without any problems in all screen modes, output directly to a monitor still works fine, too.

Made a little update video on my channel, too: https://youtu.be/9baCd5O2nG4

Thanks again for your amazing work! :)

IanSB, 2021-01-15:
@JanBeta 

>The slight flicker/sudden appearance of lines on the screen has completely disappeared (probably the timing fix you mentioned?), picture is rock stable and smooth now.

Yes there was a change to fix that. It only showed up on 1920x1080 screens and I have been mostly testing with a 1600x1200 monitor so I missed it. 

>The 625/525 switching seemed to revert back sometimes (could only reproduce it once) 

That is probably caused by the Auto Sub-profile switching.

There are two types of profile: Single profiles and Sub-Profile Sets, both stored in /profiles/Simple on the SD card.

Single profiles are stored as single .txt files and Profile Sets are stored as folders containing multiple .txt files (These are shown in the "Sub-profile" line.)

The profile sets were initially added to support auto PC mode switching from CGA to EGA etc and the profile that matches the incoming sync signals is auto selected from the files in the folder but they are also used to store both PAL and NTSC profiles for some computers like the Amiga so the correct profile will be auto loaded when connected.

Whenever the sync timing changes, the software scans the sub-profiles and reloads the correct one overwriting your temporary setting and switching on/off interlace probably caused enough of a sync timing change to trigger that. 

One workaround is to save the configuration after turning on the option as it will then re-load the saved config instead of the original but that setting will then persist until you restore default config.

Another workaround is to set "Auto Switch" from "Sub-Profile" to Off which will allow manual selection of the Sub-Profile and stop the auto loading from overwriting your setting. (Note the first press will set Auto Switch to "BBC Mode7" which is used by the BBC micro and that will make the screen go a bit crazy until you press again).

It should be possible to fix this properly by checking to see if the same profile is being reloaded again and only reload if the profile is different.