c0pperdragon/Amiga-Digital-Video #20: Feature-Request: Audio IN to HDMI

ZuiMelanieForno, 2021-01-10:
Hello.

Are all RaspberryPi Zero Pins already in Use?
If not, could two Pin´s be used to get, the Audio-Analog-Signal into the HDMI signal?

If not Analog? then maybe the DAC out pins for Left and Right?

And of course if no Pins are free on the RPi  maybe an Interrupt-based Audio signal "insertion" on the next iteration of the PCB?

I am very Far away of Electronics-Engineering but maybe it is a tiny suggestion.

 c.u. Zui from Hamburg Germany.

c0pperdragon, 2021-01-10:
As far as I know audio support is not easily possible with the way the RGBtoHDMI software works. This is basically a bare-metal program using a very tight machine code routine to sample the data from the GPIO pins. Interrupts would totally mess up everything.

ZuiMelanieForno, 2021-01-10:
@c0pperdragon OK got it, and thank You for the reply.

I saw on the Schematics of the Amiga 500 the Paula-Chip would be the Audio DAC, and it is in the proximity of the Denise-Chip.

The idea of a Interrupt based solution is not achievable. (Check or better Uncheck)

About free GPIO Pins on the RPi if I see it right there are free Pins.

Maybe a additional circuit could be added to the PCB.

I found a DAC based on the PCM5102, it needs only 5 PIN´s to connect to a RPi-Zero.

How to get a Signal from the Paula-Chip I dunno...

Is it allowed to place here a Weblink?

[https://raspberrypi.stackexchange.com/questions/76188/how-to-make-pcm5102-dac-work-on-raspberry-pi-zerow](url)

This is what I found.

Do I remember it correct that Amiga-Audio is 14bit per Channel?

I gonna try to work on it, and gonna try to get into Amiga back again.

What a Hobby...

c.u. Zui

RobSmithDev, 2021-01-11:
Might be easier (although not as clean) to add something like this to the HDMI output from the pi, and add the audio through this.

https://www.amazon.co.uk/gp/product/B01B5FR7PY

Unless theres some kind of similar IC that could do the sampling and add it to the signal I cant see the PI having time to do it without messing up the timings.  Does make me wonder whats inside this though, and weather that could be included in the adapter board somehow.

c0pperdragon, 2021-01-11:
Overall, I am quite reluctant to touch anything that has to do with audio. Opposed to a digital video solution that can actually achieve true perfection, the quality of audio so much a matter of subjective opinion.
I am absolutely sure that whatever anyone can come up with will never be good enough. 
Also converting the analog audio signal from the Paula chip to digital HDMI and later back to analog again will surely not help the quality.

c0pperdragon, 2021-01-11:
@ZuiMelanieForno
This DAC chip does not help (making the Pi output analog audio). We would need some ADC to make the Pi _input_ analog audio to add to its HDMI stream. Also it needs to work with the bare-metal solution without messing up all the (already super-tight) program timings.

ryanm101, 2021-01-14:
Might be naive here but does HDMI use separate lines for audio?  It would require an adaptor board but having something just disconnect the audio lines from the pi and provide them from a separate source (kind-of a breakout board / same way PoE injection would work with RJ45). You'd still need ADC but it would be straight from the phono connectors to the HDMI lines.

c0pperdragon, 2021-01-14:
No, HDMI provides audio data digitally and time-multiplexed with the video signal.

c0pperdragon, 2021-01-18:
Due to continuing interst in this top, I reopen this issue. 
But I can not do anything about this myself as this is completely on the software side of the RGBtoHDMI project. 
Mayme @IanSB can share his opinion on this matter.

ZuiMelanieForno, 2021-01-18:
Thank You for reopening this Issue. :)

I will suggest and ask for a "Proof of Concept" thingie at IanSB´s Project.

I installed KiCad for that Matter...

c.u. Zui

ZuiMelanieForno, 2021-01-22:
Hi, Amigians and Amigianas...

I opened a new Issue at the [RGBtoHDMI-Repository](https://github.com/hoglet67/RGBtoHDMI).

[Proof Of Concept: Audio Signals into HDMI-Output (Muxing) #177](https://github.com/hoglet67/RGBtoHDMI/issues/177)

Maybe that will help... Somehow?

c.u. Zui

ZuiMelanieForno, 2021-01-30:
Hello.

I ordered two Raspberry Pi Pico Microcontrollers.

They have 3 Pins for Analog Input.
Maybe it could work out of the Box, but i assume there will be some extra Ciruit to make an Audio-Grabber.
When (If) I can get a decent Input, I will try to couple that with the RPi Zero to see if I can get somehow a Network of I²c or such... :)

@JanBeta is willing to help, and if he can find the Time he will do a Oscilloscope Read of the Audio Pin´s.

So, stay tuned on 44.1 kHz for more Audio ;)

c.u. Zui
 
This is the same Comment like in this Isssue:
https://github.com/hoglet67/RGBtoHDMI/issues/177

Miwer, 2025-11-13:
Hi,

Sorry to bump this old topic, but I definitely see a value in incorporating audio from the Amiga into the HDMI connection, so external injectors can be avoided.

So I just want to point to the fact, that development is ongoing on the RGBtoHDMI to support audio capture. It looks like the software side of things is already working, so it might just be a hardware mod needed to incorporate audio into this adapter.
This issue was last updated earlier this year.
https://github.com/hoglet67/RGBtoHDMI/issues/376

The above issue also links to a forum post describing the details of the audio support "using the LED2 and FFOSD pins" due to lack of spare GPIO pins.
https://www.stardot.org.uk/forums/viewtopic.php?p=452912#p452912

I hope this is useful info for making audio support possible for the next revision of this adapter.

Miwer, 2025-11-13:
@c0pperdragon oh - I was not aware of that solution. Thank you for the hint. 😄