c0pperdragon/Amiga-Digital-Video #21: General discussion 2

c0pperdragon, 2021-01-15:
Continuation of https://github.com/c0pperdragon/Amiga-Digital-Video/issues/1

culexus8, 2021-01-15:
Do you see some minor flickering on wb? On my 600 it is really good in games but I see some noise arround icons/Windows/edges, I will try to do some changes with how I wire it up. Shortsen cable try a other power source just to eliminate some stuff to see if the might give any changes.

c0pperdragon, 2021-01-15:
I have just created a specific A600 thread. https://github.com/c0pperdragon/Amiga-Digital-Video/issues/22

LemaruX, 2021-01-19:
Has anyone measured how much current this draws with the Pi connected?

Pete-Cooper, 2021-01-19:
Im testing my first board (ordered from JLPCB with all the surface mount components placed) and I'm getting a flashing screen on the load workbench screen. It looks good but flashes every few seconds. If I open the menu the flashing stops. There's also a thick black line towards the bottom of the screen. Is this a setting mismatch or do I need to troubleshoot my hardware?

IanSB, 2021-01-19:
@Pete-Cooper 

Flashing means that no timing profile is matched. The download linked on the info page doesn't support NTSC so if you are running an NTSC machine that would explain it.
The latest beta does support NTSC but it is untested:
https://github.com/IanSB/RGBtoHDMI/releases/tag/3a23a05

Pete-Cooper, 2021-01-19:
@IanSB thank you! That beta did indeed fix my issue. I haven't had a chance to really test but it fixed the load workbench screen and monkey island, that is all I was able to test before work.

IanSB, 2021-01-22:
@Pete-Cooper 

That beta had the wrong aspect ratio set for NTSC profiles so here is an updated beta14 with that fixed:

https://github.com/IanSB/RGBtoHDMI/releases

bw21p25, 2021-01-27:
@IanSB Yes, that aspect ratio looks better, thank you! Ordering the bluepill to integrate FF OSD next.

ilmarkve, 2021-01-31:
When using this device, can I use Raspberry Pi Zero's TV out pins to get a composite video signal?

c0pperdragon, 2021-01-31:
I guess this should in theory be possible when the software is prepared for that.  @IanSB is the expert on that.
Maybe you could do that  by patching the config.txt onthe SDcard in some way.

Of course the big question still remains:  Why?
Composite video has such a poor quality in comparision that this seems to be totally besides the point.

ilmarkve, 2021-01-31:
I know it's besides the point but being able to replace the original monochrome composite output with a colored one would be a nice bonus.

abrugsch, 2021-02-01:
Slightly off topic, but what is the 1-chip memory expansion in [this picture](https://github.com/c0pperdragon/Amiga-Digital-Video/blob/master/installation/cabling.jpg)?

c0pperdragon, 2021-02-01:
I forgot where I got this expansion from. But it is actually working great. It is also compatible with the 1 MB chip ram modification for my particular main board.

ilmarkve, 2021-02-02:
Looks like it's from sordan.ie. They no longer have that specific model in stock and I guess it has been replaced by this: https://sordan.ie/product/637/amiga-500-500-512kb-05mb-trapdoor-ram-memory-expansion/

Nicebrook, 2021-02-17:
How do I get i touch with you c0pperdragon? I can't find your email?

c0pperdragon, 2021-02-17:
Just look at the bottom of the project main page.