c0pperdragon/Amiga-Digital-Video #22: A600 and other constrainted installation

c0pperdragon, 2021-01-15:
This thread is intended for discussions on how to get machines working with the generic small adapter board.
I don't intend to actually build such boards, as it should be fairly easy for anyone experienced enough to also do the rest of the installation. 
I encourage the use of this thread as a means of communication to help each other with difficult installs and maybe to source parts or PCBs in a more cost-effective way.

c0pperdragon, 2021-01-15:
If you have flickering pixels this may mean the color signals change too closely to the edge transistions of the clock. Try ot use a different clock source that has some phase shift.

culexus8, 2021-01-15:
Found the issue, my ground wire was to long so after shortening it the picture got stable.

c0pperdragon, 2021-01-16:
Hmm. If just varying the ground connection fixes this, I think this is now just a lucky coincidence that the signal timings match up. Next week it very well may become fluky again.
 
If you happen to have an oscilloscope, you could check the relation between your clock signal flanks and the point where the color lines change voltages.  There should be a significant time offset between these points (ideally 35ns, but anything about 15ns should also be OK):

culexus8, 2021-01-16:
Well remember that im using motherboard that is not in good shape, it needs recap and also my power supply is not god. I see 4.6v on the raspberry so there is allot to fix before I will do any check with the scope. I will orde a new quality power supply from icomp so I know I will have stable 5v and low ripple. And then I will do the install in a known good system. If I get problems then I will look more into it.

culexus8, 2021-01-17:
Testing out where to place the board and RPi zero. My Main problem is now to get a mini hdmi to hdmi that will fit inside. There is not allot of room:/
I had to drill two 2mm holes for the screws to hold the board, but is not a big deal I think. If anyone have a link for a mini hdmi male to hdmiA female that is slim and flexible please give me a hint🙂

[image]

Rakeim132, 2021-01-17:
On the A600, the memory extension A604 carries all information required as it is used for some flicker fixer cards. The adapter could be designed to fit neatly in it..?

culexus8, 2021-01-17:
You May be total correct, that would have been a sweet option to add it to ram board🙂 I need to look into that👍

culexus8, 2021-01-17:
Looks like indivison is using some other signals to produce video:/ I can`t fint the same points on the a604n. would have been a brilliant solution. But if somone sees it yeah then please tell :)

c0pperdragon, 2021-01-17:
The RGBtoHDMI solution depends on the Denise chip to generate the RGB colors for each individual pixel. As far as I can see in the schematics of the A600, this information is only passed on to the analog video circuitry, but not to the expansion port.

Eriond, 2021-01-19:
The Indivision (and possibly some other "flickerfixers") recreate the circuitry of the Denise. By listening to the Agnus-Denise communication on the RGA-bus which is exposed on the A600 memory expansion connector, and by intercepting the actual fetching of graphics data from the chip memory, you could build an equivalent (digital) graphic output from within the memexp bay. But IMHO if you'd go to all that trouble, why not make something more?

Rakeim132, 2021-01-19:
@Eriond: very interesting and totally understandable, mate.
Sorry guys, I thought it was an easier path and a good idea.

culexus8, 2021-01-24:
Some updates on the A600 ver1.5 install. 
It is now finally working 100% with the latest Beta 15 no added settings or changes to the software settings. I had some small issues with flickering in the picture before I found the error. The adapter board I use (small) have the 5V power supply directly into the GPIO and the raspberry do the voltage regulation for 3.3V back to the logic`s on the adapter board. This may have given the flickering I had issu with. So my fix was to power the raspberry pi board trough the normal usb micro port, flickering was then history :) 

[image]

I have also done some 3D printing to get the install compact and not interfere with any other expansion cards, the hdmi is right under the board where the RF-modulator normaly placed so you have a easy way of hooking it up :) Just the same stuff they use to do on any vampire install. 

[image]

[image]

[image]

[image]

[image]

I will go trough my install and make the 3D files available for download and also a step by step guide if there is interest. 
The install is not hard and is reversible if you need to. Tested on rev1.3 and rev1.5.....

Thank you c0pperdragon and IanSB for this amazing stuff :)

c0pperdragon, 2021-01-24:
That is a pretty neat install. Congratulations!

One thing still bothers me a bit: To get rid of the flickery pixels, you basically just fiddled around with the power supply and ground a bit. I don't think this was the true cause of the problem, but it just now seems to work more or less by accident.

If you want to provide a stable solution that will reliably work for other users as well, I would suggest to do some measurements first. If you have an oscilloscope, you really should check in which relation the edges of the clock signal are to the edges of the color signals.

culexus8, 2021-01-24:
I do have a scope but I do not understand to much of it hehe. I will contact you by e-mail if it is okay and try to find a solution.

seb132, 2021-01-26:
@**culexu8**: nice job, really... Do you have all ECS modes working ?
@c0pperdragon: could it be possible to have in the description the list of the original video modes that are managed without issue?
I'm struggling to understand the current limitations with ECS, if any...

c0pperdragon, 2021-01-26:
I added an explanation of the OCS screen modes to the documentation. 
Note that this restriction applies to all adapter versions - it is a fundamental limitation of this Raspberry Pi - solution.

culexus8, 2021-01-26:
Seb132: thanx🙂 I have not teste allot but I do use high res laced pal on the wb and all the standard games do work good so far. Can't find any issue with this solution. My biggest problem must be that my furia is not as stable as I was hoping for. But this have no thing to do with the hdmi install🙂

sonic2k16, 2021-02-01:
How many layers is the PCB...

culexus8, 2021-02-01:
> How many layers is the PCB...

Ismt is two layers.

Sebretro, 2021-02-08:
Hi I would really like a small guide and 3d files if you will share. I am waiting for the boards from C0pperdragon and are now trying to find the components in my neck of the woods.

sonic2k16, 2021-02-08:
How can I through KiCad get MPN from a project (manufacturer part number = component identification code) from a CSV file (BOM).
Thank you.

samowitsch, 2021-02-10:
Hi. I have a Minimig an wanna try to get RGB2HDMI working with it. In the picture below you can see the RGB locations as well as the HSYNC and VSYNC. Can i use theoretically HSYNC for SYNC and VSYNC for CLOCK? I am a little bit confused about different naming.
THX and Regards
[image]

c0pperdragon, 2021-02-10:
For the standard Pi software to work you need a CSYNC input (which is a kind of combination of HSYNC and VSYNC). I don't know if this is available on the minimig somewhere.  Otherwise the Pi software needs to be modified. 
Also you must have some system clock (7.1 MHz) to feed into the adapter to correctly sample the pixels.

samowitsch, 2021-02-11:
@c0pperdragon thx for info. i will investigate. Thats the signals that are coming directly out of the FPGA of the Minimig. There is no CSYNC. A system clock should be some where at the board.

IanSB, 2021-02-11:
@samowitsch 
I just had a look at the minimig rev 1 schematic and you can certainly use the RGB bits and Hync (which is composite sync in 15Khz mode) but you would also need a pixel clock signal which probably only exists internally in the FPGA  (It's actually pixel clock divided by 2 so 7 Mhz for a 14 Mhz pixel clock etc.). 

If you can modify the FPGA code you could change it to output the pixel clock on the Vsync pin.

Other than that you could use the CPLD version of RGBtoHDMI as that only requires the 12 bits and sync because it re-creates the pixel clock in the CPLD.

samowitsch, 2021-02-12:
> If you can modify the FPGA code you could change it to output the pixel clock on the Vsync pin.

@IanSB Thats far beyond my skills ;o)

culexus8, 2021-02-15:
> Hi I would really like a small guide and 3d files if you will share. I am waiting for the boards from C0pperdragon and are now trying to find the components in my neck of the woods.

Hi, yes I will free the 3D files when I get the time to finish them and they are at a final stand point. Not shure where I will upload them but I will update here.

solarmon, 2021-02-17:
Hi, a question regarding the connection diagram. For CLK it shows it to take it from pin 2 of (unpopulated) C201. Is this not yhe 7Mhz signal? Looking at the board schematic I thought CLK should be taken from the _CDAC signal? Or am I reading it wrong?

c0pperdragon, 2021-02-17:
The CDAC in the schematics is actually a misnomer. On the silk screen, I have re-named it to the generic term CLK. Into this you need to feed either ther CDAC signal when running with a OCS Denise, or the 7Mhz signal when running with a SuperDenise.
For a A600 taking the 7Mhz is just fine. In this case you also need to close the solder bridge JPIN1 to get the correct polarity of the clock signal.

solarmon, 2021-02-17:
@c0pperdragon I'm looking to make a RGBtoHDMI board, based on your small board, that fits directly on Denise with a PLCC socket.

In that case, then should I just do away with JPINV1 jumper (and connect directly to +3.3V); and connect the 7M signal instead of CDAC?

ryanm101, 2021-02-18:
Works for Rev 1.1 (A300 June Bug) as well

> [image]

samowitsch, 2021-02-18:
> Other than that you could use the CPLD version of RGBtoHDMI as that only requires the 12 bits and sync because it re-creates the pixel clock in the CPLD.

@IanSB you mean this thingy here?: https://github.com/hoglet67/RGBtoHDMI/wiki/Bill-of-Materials-%2812-Bit-Board%29

solarmon, 2021-02-19:
@c0pperdragon If I remove JPINV1 and connect directly to +3.3V, then is R2 and the connection to GND is really needed? Does it really serve any purpose in this situation? If I can remove it then that will reduce the component count.

[image]

c0pperdragon, 2021-02-20:
You would not need the pull-down resistor in this case.

solarmon, 2021-02-20:
@c0pperdragon Thank you for confirming that - that is what I thought.

I'm currently working on this Denise PLCC socket based board. I still need to test and confirm the PLCC-68 socket and pinouts.

[image]

The Pi would then sit on the board as such:

[image]

solarmon, 2021-02-20:
Looks like I made a major mistake. I used a PLCC-68 footprint, instead of a PLCC-52 one. What an idiot! 🙄

culexus8, 2021-02-20:
> @c0pperdragon Thank you for confirming that - that is what I thought.
> 
> I'm currently working on this Denise PLCC socket based board. I still need to test and confirm the PLCC-68 socket and pinouts.
> 
> [image]
> 
> The Pi would then sit on the board as such:
> 
> [image]

Is this for the A600? looks like a nice design :)

solarmon, 2021-02-20:
@culexus8 Yes, it is meant to fit over the A600 Denise PLCC chip.

I've now corrected the design to use a PLCC-52 socket:

[image]

However, it will probably spoil with any memory and CPU board, depending on its design.

But until I get a test build, I don't know whether it will spoil with any other A600 motherboard components.

nyteshade, 2021-02-24:
Any ideas if the constrained installation will be needed in the A3000 or if the regular RGB2HDMI solution would work there?

nyteshade, 2021-02-24:
What about the A1000?

Sebretro, 2021-02-24:
HI. How do I get a parts list from the files? I can find the IC numbers in the pdf and also the resistance and capacitance values of the R And C components but not voltage ratings and form factors/size if that is an issue. This is my first time following a project in git hub so please excuse if the question is stupid. But I have done  some soldering, fixing stuff. And also recapped my A600 so I am not completely lost.

c0pperdragon, 2021-02-26:
I don't have a specific parts list. Just use the ICs in a SOIC package (the largest SMD there is), and all passives are 0805 (imperial). Voltage rating is basically irrelevant when it is at least 10V for the capacitors.

ryanm101, 2021-02-26:
@Sebretro 

10.0 | 74LVC86AD,112 | NEXPERIA | NEXPERIA  Logic IC, XOR (Exclusive OR), Quad, 2 Inputs, 14 Pins, SOIC, 74LVC86
-- | -- | -- | --
20.0 | 74LVC574AD,118 | NEXPERIA | NEXPERIA  Flip-Flop, 74LVC574A, D, 17 ns, 200 MHz, SOIC
10.0 | CGA4J1X5R1C106K125AC | TDK | TDK  SMD Multilayer Ceramic Capacitor, 10 µF, 16 V, 0805 [2012 Metric], ± 10%, X5R, CGA Series
30.0 | 08051C104KAT2A | AVX | AVX  SMD Multilayer Ceramic Capacitor, 0.1 µF, 100 V, 0805 [2012 Metric], ± 10%, X7R
10.0 | MCWR08X3301FTL | MULTICOMP PRO | MULTICOMP PRO  SMD Chip Resistor, 3.3 kohm, ± 1%, 125 mW, 0805 [2012 Metric], Thick Film, General Purpose
10.0 | WR08X1002FTL | WALSIN | WALSIN  SMD Chip Resistor, 10 kohm, ± 1%, 125 mW, 0805 [2012 Metric], Thick Film, General Purpose

Sebretro, 2021-02-26:
Thanks for the info. I have a slight problem I can only find these 574 locally here in Sweden. 
SN74HC574DWR 28Mhz
74HC574D 133 MHz

Will any off them work or do I need the lvc version? 
Everything else was available and to buy just this chip from mouser, or such, will have a silly overhead cost. Any pointers for where I can get it from a EU online shop? Thank you all.

c0pperdragon, 2021-02-27:
These HC parts are not the correct ones, as they can not work as voltage level shifters.
Instead of LVC you could use LCX as they have very similar characteristics. But I guess these are equally hard to get.

ryanm101, 2021-02-27:
@c0pperdragon what is the risk there I tested these and they worked on mine?

ryanm101, 2021-03-01:
@Sebretro if @c0pperdragon doesn't mind / have any, i've some extra so if you are just looking 1-2 I could help you out.. you can email me at (pvtryan 100) (-at-) gmail com.

Sebretro, 2021-03-05:
Oh to bad didn't se your reply untill now. Somehow the email notification switched off. I already ordered some from the US. Gave me a reason to fill upp the order with a hot air rework station. Just to get free shipping. 😀. Your offer is highly appreciated. In a few days I will also have some extra lvc574 so same offer from me to anyone in Sweden or Europe.

krug3rf, 2021-03-16:
> Some updates on the A600 ver1.5 install.
> It is now finally working 100% with the latest Beta 15 no added settings or changes to the software settings. I had some small issues with flickering in the picture before I found the error. The adapter board I use (small) have the 5V power supply directly into the GPIO and the raspberry do the voltage regulation for 3.3V back to the logic`s on the adapter board. This may have given the flickering I had issu with. So my fix was to power the raspberry pi board trough the normal usb micro port, flickering was then history :)
> 
> 
> I will go trough my install and make the 3D files available for download and also a step by step guide if there is interest.
> The install is not hard and is reversible if you need to. Tested on rev1.3 and rev1.5.....
> 
> Thank you c0pperdragon and IanSB for this amazing stuff :)

Hi @culexus8 
on rev1.5 the soldering schema is the same like rev1.3 ?

culexus8, 2021-03-17:
It is, there is no difference from 1.3 to 1.5 so Just follow the same steps.  :)

seb132, 2021-03-17:
@**solarmon**: no chance that your design works with ANY A600 accelerator, unfortunately. 
Even any Fast Ram board I'm afraid as most of them clip on the 68000.

krug3rf, 2021-03-17:
> It is, there is no difference from 1.3 to 1.5 so Just follow the same steps. :)

thx
Ok so the 3.3v for the Pi0 is not mandatory and we must use the 5v from R406 directly connected to the RGBtoiHDMI PCB.

solarmon, 2021-03-17:
> @**solarmon**: no chance that your design works with ANY A600 accelerator, unfortunately.
> Even any Fast Ram board I'm afraid as most of them clip on the 68000.

There is one that I'm aware of that should be compatible - the A608:

[image]

But yes, most of the others will not be suitable. This is more of a learning exercise for me.

culexus8, 2021-03-17:
> > It is, there is no difference from 1.3 to 1.5 so Just follow the same steps. :)
> 
> thx
> Ok so the 3.3v for the Pi0 is not mandatory and we must use the 5v from R406 directly connected to the RGBtoiHDMI PCB.

You can power the board just like you want :) you do not need to use the 5v from the R406, you can take 5V from some other place also. It is up to you. I even powered my board with an usb cable from the floppy connector. the 3.3 will be taken from the Pi0.

solarmon, 2021-03-17:
> > It is, there is no difference from 1.3 to 1.5 so Just follow the same steps. :)
> 
> thx
> Ok so the 3.3v for the Pi0 is not mandatory and we must use the 5v from R406 directly connected to the RGBtoiHDMI PCB.

@culexus8 

The RGBtoHDMI board needs to fed 5V so that it can feed the Pi0. There is then a 5V to 3.3V regulator on the Pi0 which feeds the components on it. Asking  "3.3V for the Pi0 is not mandatory" does not make sense.

culexus8, 2021-03-17:
> There is one that I'm aware of that should be compatible - the A608:
>
> [image]
>
> But yes, most of the others will not be suitable. This is more of a learning exercise for me.

I might test this one day, :) just need some more ram and will be a perfect option. 
https://github.com/PR77/A600_ACCEL_RAM

krug3rf, 2021-03-22:
Thanks all for tour help 👍
[image]

Osmund, 2021-03-24:
> Thanks all for your help 👍
> 
> [image]
> [image]
> [image]

Looks great. Is the stl file available for download?

ryanm101, 2021-03-24:
Has there been a change? https://github.com/c0pperdragon/Amiga-Digital-Video/blob/master/smallkit/README.md mentions 3k3 res for R1/R2 but schematics say R2 should be 10K.

culexus8, 2021-03-24:
> Has there been a change? https://github.com/c0pperdragon/Amiga-Digital-Video/blob/master/smallkit/README.md mentions 3k3 res for R1/R2 but schematics say R2 should be 10K.

I use 3k3 for R1 and 10K for R2 as the schematics say.

[63 more replies]