c0pperdragon/Amiga-Digital-Video #23: Audio output on  - Please reopen issue #20

Crushless, 2021-01-16:
First, thank you for this cool project, and all the effort you put into, but please consider to reopen issue #20 (Audio IN to HDMI).
Yes, I know the code is bare metal and already tight, but probably there is just enough time to read some samples from an i2c connected ADC in the horizontal and vertical blanking intervals and insert them in the HDMI stream. Audio quality should not be such an big issue, because the sample rate has only to be slightly higher than double the output rate of Paula (see Nyquist–Shannon sampling theorem) to get decent quality.
Relaying on an external solution (HDMI audio inserter) is both inelegant and cumbersome, when the Raspberry Pi just could just insert the audio itself in the first place.

ZuiMelanieForno, 2021-01-18:
Hi @all

I maybe missed to explain the Request in the issue #20.

What I assumed is to mix the Audio (digital signal´s) from the **Paula Chip (Pin 30 for the Right Channel, Pin 31 for the Left Channel)** into the HDMI-Output-Signal.

What I saw are some free Pin´s on the Raspberry Pi Zero GPIO Pin´s, which are not connected,
and could be used for that Purpose.

If needed maybe extending the PCB to some more Logic (IC² ?) or do a secondary PCB with the same Passthru idea of using
an in between Socket.

I will try to get my Amiga 2000 to run again, and mybe I can get an Oscilloscope rented,
to get an Idea of the Paula-Chip Pin´s 30 and 31 Signals.

Like Paul Shannon was Juggling in his Past (I was a Juggler in my younger Days), we should try to Juggle in the Audio, it is just a case of Cascading... 🔢 

Please dear @c0pperdragon reconsider a "Audio Signal Muxing".

c0pperdragon, 2021-01-18:
Reopened #20

closed 2021-01-18 by c0pperdragon