c0pperdragon/Amiga-Digital-Video #24: I need the BOM

UnaClocker, 2021-01-17:
I can't seem to find the BOM for the Denise adapter board. I downloaded the gerbers, and had a set of boards made at OSHPark. Got a BOM file for those of us soldering the thing together ourselves? Maybe I'm missing it.

Tom-PCB, 2021-01-17:
[amiga_bom.xlsx](https://github.com/c0pperdragon/Amiga-Digital-Video/files/5825610/amiga_bom.xlsx)
[amiga_cpl.xlsx](https://github.com/c0pperdragon/Amiga-Digital-Video/files/5825611/amiga_cpl.xlsx)

files are made for "JLCPCB"  to maufacture the PCB and solder the SMD parts

https://cart.jlcpcb.com/quote?orderType=1

c68696c66, 2021-01-17:
The V2 bom is available, see https://github.com/c0pperdragon/Amiga-Digital-Video/issues/15#issuecomment-761044838 
-> https://github.com/c0pperdragon/RGBtoHDMI/blob/master/kicad_AmigaAdapter/V2/deniseadapter_bom.ods

ryanm101, 2021-01-20:
@c0pperdragon what is the BOM for the small board? (mainly after the footprint of the resistors / caps)

c0pperdragon, 2021-01-20:
The passives are all 0805.

closed 2021-01-26 by c0pperdragon