c0pperdragon/Amiga-Digital-Video #27: Feature request - Graffiti emulation

rzookol, 2021-01-23:
Would be nice to add http://wiki.icomp.de/wiki/Graffiti emulation.

c0pperdragon, 2021-02-02:
As the solution here is just using the output of the Denise chip itself, here is no way to support different graphics cards.

rzookol, 2021-02-03:
But graffiti isn't gfx card, it's is just converter of ~chunky data sent to amiga video port to color output. Similar to DCTV card earlier. So we would get C2P for free, and 256 colors in 320x256. Denise output should be ok as both cards (DCTV and Grafitti needs only 23pins Video port of A500/CDTV/A1200).

c0pperdragon, 2021-02-03:
Ah, OK, so I misunderstood the issue.

seb132, 2021-02-24:
I have a Grafitti, it's a nice card but limted range of applications (even Shapeshifter is not easy to setup).
I'd say priority is elsewhe, but agree with "nice to have". :)

aiobofh, 2021-04-16:
Right! Just putting "ColorMaster 24" here too... If that provides more useful modes.