c0pperdragon/Amiga-Digital-Video #29: where to buy TLV74333PDBVR

TheAmigaman, 2021-01-30:
Seems there's a bit of a shortage of TLV74333PDBVR.
Most suppliers seem to have none and are quoting 16 weeks lead time.
The few that have them are quoting daft postage and packaging unless I buy a 1000 or more.

Are there any alternatives ?

abrugsch, 2021-02-01:
The footprint is pretty consistent between manufacturers (known as a jellybean part) so as long as the package is the same (SOT23-5) and has sufficient output current (TLV743333PDBVR can drive 300 mA - but I haven't checked yet how close to that capacity is needed) then you can pretty much pick any other 3.3v LDO from your preferred vendor through a filtered search. (and then check agains the datasheet to make sure the chosen one IS pin compatible)

so for example, here's a suitable filtered search on mouser:[clicky](https://www.mouser.co.uk/Semiconductors/Power-Management-ICs/Voltage-Regulators-Voltage-Controllers/LDO-Voltage-Regulators/_/N-5cgac?P=1z0wa2eZ1z0z7vbZ1z0z33qZ1z0wa3yZ1z0wbyxZ1z0w877Z1z0wd8oZ1z0wd73)
and I've picked out a few examples that are pin compatible and have sufficient power output*

*I went down to 250mA but that may not have been appropriate. I will check but there are sufficient examples of >300mA LDO's that are available - the maximum rating for the logic IC's are 100mA each so probably safest to err on the side of caution. I haven't done any experimentation to see how wildly that varies... so probably best to avoit <300mA parts

TPS7A20 - same power output (300mA) ultra low noise
NCP163ASN330T1G - slightly lower output (250mA)
TLV73333PQDBVRQ1 - automotive spec variant otherwise same as otiginal part
TLV75533PDBVR - 500mA higher current capability
MCP1812BT-033/OT 300mA
MCP1812AT-033/OT 300mA
TLV70333DBVR - 300mA
LDK130M33RY - 300mA

now some of these have other differences that may or may not be relevant like automotive grade which will make it more expensive but otherwise un-necessary for use in an Amiga

tyristori, 2021-02-17:
I use AMS1117-3.3 in mine and it works fine. Output current is 1A.

Jukka-M, 2021-02-27:
Is this compatible. ( SGM2019-3.3YN5G ).
https://www.aliexpress.com/item/4000163877756.html?spm=a2g0o.cart.0.0.101f3c00UCzlyh&mp=1

abrugsch, 2021-02-27:
Datasheet here https://datasheet.lcsc.com/szlcsc/1809051429_SGMICRO-SGM2019-3-3YN5G-TR_C23613.pdf seems to suggest so. 300ma, 3.3v and correct pinout 👍

Jukka-M, 2021-02-27:
Thanks for the quick reply.
I ordered this: https://www.aliexpress.com/item/4000163877756.html?spm=a2g0s.9042311.0.0.47584c4d8VD7E9

tyristori, 2021-03-06:
I'm wondering what's the purpose of 3.3V regulator on A500 board as you can get 3.3V from Pi?

Small adapter variant doesn't have the regulator and takes 3.3V from Pi.

InfiniteExpression, 2021-03-15:
> I'm wondering what's the purpose of 3.3V regulator on A500 board as you can get 3.3V from Pi?
> 
> Small adapter variant doesn't have the regulator and takes 3.3V from Pi.

I was just going to ask that (although disappointed not to see an answer for you already) I'm currently building V2 for an A500, and the 74333 regulator is "unobtanium" from any USA source. Is there any reason not to use the 3.3 from pin 1 of the PI?

LinuxJedi, 2021-03-15:
> I'm wondering what's the purpose of 3.3V regulator on A500 board as you can get 3.3V from Pi?
> 
> Small adapter variant doesn't have the regulator and takes 3.3V from Pi.

You can technically do this as there should be enough current available from the regulator, but it isn't recommended to pull more than 50mA from any GPIO pin (including 3.3v pins). If you are worried about the cost then change it to a 1117 SOT223 setup in the schematic (you'll likely need to change C4 too and possibly add an input cap), will likely be cheaper and easier to hand solder.

mthiesen, 2021-03-26:
I just finished building my V2 for use with my Amiga 500 and I used the [AP139-33WG-7](https://www.digchip.com/datasheets/download_datasheet.php?id=4602397&part-number=AP139-33WG-7) as a drop-in replacement for the TLV74333PDBVR. It seems to be working just fine.

I chose this one because it was one of the few I could find that is available, pin-compatible and does not require an input cap.

FreezeyBeans, 2021-04-12:
Slightly related on the topic of the BOM - I also notice that the 74LVC574 are available in two frequencies from vendors, namely a 150MHz and a 200MHz. - Is there any preference on which to use for the v2 adaptor boards for self builders?

LinuxJedi, 2021-04-12:
> Slightly related on the topic of the BOM - I also notice that the 74LVC574 are available in two frequencies from vendors, namely a 150MHz and a 200MHz. - Is there any preference on which to use for the v2 adaptor boards for self builders?

I've used the TI 150MHz ones and they appear to work quite well. I have some of the NXP 200MHz but haven't tried them in this board yet.