c0pperdragon/Amiga-Digital-Video #30: DIY board troubleshooting

echolevel, 2021-01-30:
Thanks very much for all the hard work on this project! I got some V2 PCBs printed and I assembled one today, but I'm just getting a black screen on HDMI and I'm wondering if somebody can help me troubleshoot it.

Some notes:
* the board is currently in an A500+, with the jumper set for Super Denise
* the RGB signal to my 1084S CRT monitor is fine
* all traces between the Denise and the board components, and between the board components and the Pi, seem to have good continuity
* as described in the README, I've unzipped the software onto the root of a freshly formatted microSD
* the Pi Zero quite happily boots and outputs the proper HDMI signal when I insert a Raspbian microSD, so I know that the Pi is getting the right voltage and booting properly
* the LCD monitor that's connected to the Pi does detect a signal from the Pi when it's running RGBtoHDMI; the backlight illuminates, but it just displays a black screen
* if I fire up Deluxe Paint on the A500+ in 2 colour mode, attach an oscilloscope to GND and B0 on the Super Denise, and then adjust the Blue level of the background colour (white) in DP's palette, I get commensurate high/low average voltages on the scope
* if I run a GPIO monitor from the command line when running Raspbian, I can read individual pin states while the Amiga is running...but when doing the above DP test I see no meaningful  fluctuation in B0, B1, B2, B3

Can anyone think of anything else I might test to troubleshoot this? I don't know much about diagnosing stuff with the scope, especially when it comes to video signals, so I'm not sure if any of the above is valuable information. Is there any config option tweaking I can do on the Pi side to output a test signal?

(One more thing: just ONCE I saw the multicoloured test screen when the Pi with RGBtoHDMI started, but I think it was after switching from the Rasbian to the RGBtoHDMI microSD so perhaps there was some kind of bootloader cache thing going on)

Thanks in advance!

tk-amiga, 2021-01-31:
Hi,  I've soldered few of these too and i did one a bit too hastily. At first there was a short and i got no image, but it was Amiga that was not booting at all. I had also the normal Amiga RGB output attached to same tv so i noticed that quite fast. You get image from the rgb at all?
And if it's a first boot for the sdcard it will take a bit longer for it to configure itself for the first time. 
If you have the button attached to the adapter can you get the menu open with it?

echolevel, 2021-01-31:
Good shout, thanks - I'll try attaching a button and see if I can see the menu! Yes, regular Amiga RGB output is fine - that's being displayed on the CRT, whereas the HDMI is going to an LCD monitor.

Also interesting to know it autoconfigures once, rather than on every boot. Which file does it create/modify? config.txt?

echolevel, 2021-01-31:
Hmm, I can't get the onscreen menu to show up. I don't have a button, so I'm shorting the button pins for varying lengths of time with no success.

echolevel, 2021-01-31:
Okay, I think the XOR gate is fried (and my spare) - I must have used too much hot air on them.

I'll order some more, but meanwhile: can anyone tell me if the menu should show up even if the Pi isn't receiving the correct signals from the Super Denise via the ICs? If so, then I've got a second problem to troubleshoot...

IanSB, 2021-01-31:
@echolevel 

>Okay, I think the XOR gate is fried
>can anyone tell me if the menu should show up even if the Pi isn't receiving the correct signals 

The menu button will not respond if the Pi zero isn't receiving the pixel clock which is generated by the XOR gates.
If you have a 5v TTL USB serial adapter you can hook that up to the debug serial port on the Pi zero to confirm it is booting up.
Serial out is on pin 8 and ground is on pin 6 @ 115200 baud

echolevel, 2021-01-31:
That's reassuring, thanks Ian! I have a USB serial adapter lying around somewhere so I'll check that, but I'm now pretty confident that the Pi is booting fine and it's just my ICs that are causing problems.

Cheers!

de-nugan, 2021-02-14:
I had the same problem initially but in my case the RPi wasn't booting because the SD card was partitioned using GPT instead of MBR. The LED on the RPi did not light with the GPT card inserted. Repeating the setup with an MBR-partitioned card worked fine.

closed 2021-02-20 by c0pperdragon

echolevel, 2021-02-20:
Thanks for closing, @c0pperdragon, and for your hard work on this project - finally got it working :) My 3rd PCB eventually tested successfully with an Arduino Nano, but wouldn't allow any direct passthrough of Denise on A500 or A500+. Finally I tracked the fault down to a short on the XOR - reflowed the XOR and all is well!