c0pperdragon/Amiga-Digital-Video #32: Reason for R1

hansliss, 2021-02-03:
I've searched the code for references to GPIO pin 18, and as far as I can see it's never set as an input pin. It's used to signal something to the adapter card. So why is R1 there on the Denise adapter? All it does is to pull this output pin to ground, right?

I seem to have used a far too small resistor by mistake, so I'm wondering if I can simply remove it.

IanSB, 2021-02-03:
No, you can't remove the R1 resistor on GPIO18 because it is used to detect the Amiga board (Simple mode). If it is missing the software assumes it is running on the CPLD board instead. The relevant code is in init_hardware() which sets GPIO18 (VERSION_PIN) to an input and reads back the level. If the resistor is fitted it reads back as low, if not it reads back as high due to a weak internal pullup on the Pi. It is then set as an output to control the CPLD board.

hansliss, 2021-02-03:
> No, you can't remove the R1 resistor on GPIO18 because it is used to detect the Amiga board (Simple mode). If it is missing the software assumes it is running on the CPLD board instead. The relevant [...]

Ah, ok. Thanks for the info! It turns out I didn't use the wrong resistor after all, anyway.

closed 2021-02-05 by hansliss