c0pperdragon/Amiga-Digital-Video #33: Video problems

cgoffa1, 2021-02-04:
Just installed everything today and I am very impressed.

However sometimes on start up, the image will be shown in a red hue.  Other times the colors will be correct but some pixels will be flickering side to side (mostly in the middle of the screen).  And other times the picture is perfect.

Any idea how to remedy the two problems?

IanSB, 2021-02-04:
Could you post some screencaps of the problem images (short press on button, files stored on SD card)

c0pperdragon, 2021-02-05:
Which adapter version do you use? Did you build one yourself, or is this a read-made one?  
Did you check if the jumper at the underside (if it is a Rev 2 adapter) matches your Denise variant?

cgoffa1, 2021-02-05:
I have the first version of the board.  I bought it pre made and for reference it is installed in an Amiga 500 rev 5.  

I will post screenshots as soon as possible

cgoffa1, 2021-02-05:
[image]

So today the red hue issue was not duplicated.  The flickering pixels continued but I was unable to get them in a screenshot.  Then all of the sudden the thing went haywire and the screen was rolling vertically and acting like it was losing sync.  The screen shot attached kind of shows the distortion.

cgoffa1, 2021-02-05:
I deleted the contents of the sd card and reapplied the firmware but this did not remedy the problem.  For reference I am using the latest release of the firmware.

c0pperdragon, 2021-02-05:
Maybe there is indeed something faulty with your adapter. When you have a Rev.1 this was probably not made by me. Did you already contact the person who built it?  Maybe there is a bad solder joint on one of the ICs?

cgoffa1, 2021-02-05:
I am beginning to agree with you. The build quality of the board is not that good. How can I order one from you?

c0pperdragon, 2021-02-05:
Drop me an e-mail as explained on the bottom of the project main page.  But be aware that there is quite a long waiting list right now.

IanSB, 2021-02-06:
@cgoffa1 

The loss of sync and random colour changes do sound like an intermittent contact on one or more of the chips.
To investigate this, you could try gently nudging the legs of the ICs with a pin or needle and anyone that moves will likely be the problem. You might be able to fix it by pressing on any such leg with the clean tip of a soldering iron to remelt any solder but I wouldn't add any new solder as you will likely end up creating a short unless you are experienced with surface mount soldering.
The sync pulses are routed via U2 so that would be a good starting point.

closed 2021-04-16 by c0pperdragon