c0pperdragon/Amiga-Digital-Video #35: A3000 Support?

nyteshade, 2021-02-05:
Is the new ECS mode support working for the A3000?

c0pperdragon, 2021-02-20:
Sorry, but no. The current solution is restricted to the 70ns / pixel timing. No super-high-res or productivity modes.

closed 2021-02-28 by c0pperdragon

jlavinen, 2021-04-18:
What prevents implementing 35ns pixels? Pixel clock not available? New hardware needed to do this? RP0 too slow?

c0pperdragon, 2021-04-19:
Pixel clock could probably be created by combining both the 7Mhz and the CDAC line. This would require a new adapter at least.
But the true reason is the speed. The Pi is not fast enough to transfer this amount of data through the GPIO pins using the CPU.
It is even barely fast enough for the 70ns modes. 
Using other versions of the Pi will also not help, as having multiple cores only makes this kind of real-time processing even more difficult.