c0pperdragon/Amiga-Digital-Video #36: A1000 space constraint

hansliss, 2021-02-12:
The Denise adapter would fit nicely in an A1000, but there's no chance to fit a Raspberry Pi Zero there, since Denise is just next to the PSU. Would it be likely to work if I connect the Pi using a ribbon cable, and place it somewhere else, or would that introduce noise that would ruin the quality?

I'd need to do some motherboard mods for this to create the /CDAC signal and bring that and /CSY to the Denise socket, build a new adapter with a pin header for the Pi, as well as making a suitable ribbon cable, so I prefer to check first whether it's worth it.

c0pperdragon, 2021-02-12:
When the cable is not too long, this could probably work. But you need to find a way to connect it so that the pins are not mixed up. You probably have to solder the pins of the Raspberry Pi upside down.

hansliss, 2021-02-12:
> When the cable is not too long, this could probably work. But you need to find a way to connect it so that the pins are not mixed up. You probably have to solder the pins of the Raspberry Pi upside down.

Yep, you figured that out quickly. It took much longer for me. :) I ended up using individual Dupont cables instead, with non-optimal result. See #28

kipper2k, 2021-02-15:
I have designed an adapter board for the A1000 which doesn't interfere with the PSU and has needed headers for you to attach dupont headers as needed. I will show it off tomorrow

kipper2k, 2021-02-15:
here is a picture of my a1000 adapter board, It is untested, it should work as i modified my A600 board for the A1000

[image]

here are the gerber files. As i said i have not tested this to please use at your own risk.

[a1000hdmiadapter.zip](https://github.com/c0pperdragon/Amiga-Digital-Video/files/5979117/a1000hdmiadapter.zip)

hansliss, 2021-02-15:
Excellent! Thanks! I will include this in my upcoming PCB order.

c0pperdragon, 2021-02-15:
When you design a board for the A1000 and intend to connect the Pi from it with a ribbon cable, you should include as many GND wires  in the cable as possible. I didn't care too much for my Rev.2 adapter because the GNDs are connected through multiple pins already. On a ribbon cable on the other hand, you need to be much more careful when transmitting power or coupling to GND.

kipper2k, 2021-02-15:
here is my a1000 board layout, it should fit in both Pal/NTSC machines. I had to offset the pi header so the daughter board on the NTSC machine didn't interfere. Please give the layout a good checking to make sure i didn't mess up.

[image]

here are the Gerber files.

[a1k001.zip.zip](https://github.com/c0pperdragon/Amiga-Digital-Video/files/5984433/a1k001.zip.zip)

hansliss, 2021-02-16:
Nice work! Thanks! I'll include this in the next PCB order and give it a go.

IanSB, 2021-02-16:
@kipper2k 
@hansliss 

If you still get flickering pixels on some edge transitions as noted in https://github.com/c0pperdragon/Amiga-Digital-Video/issues/28 then try fitting 74VHC parts instead of 74LVC parts. 

I had a similar problem with my rev 5 A500 and looking at the 7Mhz and _CDAC lines they had lots of ringing on them which wasn't present on rev 6 or later. Looking at the rev 6 circuit diagram there is some filtering on those lines using a 47R resistor and 47pF capacitor but those components are missing from rev 5 and presumably earlier boards.

After some experimentation I found that fitting the slower 74VHC parts fixed it for my rev 5. Note this is 74VHC not 74HC as the 'V' means they are 5v tolerant when run from lower voltages which isn't the case with HC parts.

abrugsch, 2021-02-16:
@IanSB do you think it might be suitable to add the RC filter to a Rev5? I have a Rev5 here that I was going to add one to and all my ordered parts are 74LVC (is this something that's likely to afflict all Rev5's or just a few?)

IanSB, 2021-02-16:
@abrugsch 
>is this something that's likely to afflict all Rev5's or just a few

I don't know, I've only seen it on my own rev5 so far and there haven't been any other confirmed reports but the results were similar to those shown on the A1000 tests. It only seems to happen on some types of screen content like transitions from intensity 7 to intensity 8 which results in all 4 bits of that R/G/B signal changing state.

I have built up quite a few boards and I tried them all in my rev 5 and the results were quite variable from really bad to barely noticeable at all so it depends to some extent on the luck of the draw with the LVC parts.

>do you think it might be suitable to add the RC filter to a Rev5?

I did try filtering the signals going into the LVC parts on the adapter and although it improved things it didn't completely fix it so I guess the signals would have to be filtered on the main board before they go into the Denise chip and I didn't want to start cutting up the main board so I never tried that but it's certainly worth trying if you are OK with cutting tracks on your rev 5.

abrugsch, 2021-02-16:
@IanSB 
> so I guess the signals would have to be filtered on the main board before they go into the Denise chip and I didn't want to start cutting up the main board so I never tried that but it's certainly worth trying if you are OK with cutting tracks on your rev 5.
Ah yes that was the sort of mod I was talking about. can you link (or snip) the part of the schematic and I'll see if it's viable. (I'm fine with that sort of mod TBH...)

IanSB, 2021-02-16:
@abrugsch 
>can you link (or snip) the part of the schematic and I'll see if it's viable

https://www.amigawiki.org/dnl/schematics/A500_R6.pdf
If you look at page 3 of this rev 6 schematic there are 47R resistors in the _CDAC and 7Mhz lines from the Agnus and 47pf to ground on the _CDAC line plus on page 6 there is a 47pf trimmer on the 7Mhz line. The CCK line  and CCKQ line are similarly filtered and probably would need that to keep everything in phase. I haven't found a rev5 schematic but don't see those components on my rev 5 board.

>do you think it's worth making HC parts the standard in the BOM? 

They would have to be tested on later revs to make sure they didn't cause any issues with them. Using the slower parts may reduce the time available to read valid data on each pixel clock cycle. (BTW VHC, not HC)

SirCathal, 2021-03-14:
I made some progress with my Pal A1000
https://github.com/c0pperdragon/Amiga-Digital-Video/issues/28#issuecomment-798964142