c0pperdragon/Amiga-Digital-Video #37: A600 Amiga HDMI video design files

kipper2k, 2021-02-17:


kipper2k, 2021-02-17:
Here is my a600 design, tested working fine. It consists of 2 boards and an FFC cable to connect them together. 
[image]

[image]

[image]

[a600design_gerbers.zip](https://github.com/c0pperdragon/Amiga-Digital-Video/files/5997607/a600design_gerbers.zip)

You can see the small square board sits around the DAC and you simply solder to the pins. This board is connected to the adapter board by a 20pin 0.5mm FFC cable.

abrugsch, 2021-02-18:
Excellent! My A600 is looking forward to this

c0pperdragon, 2021-02-20:
Wow!  This is an ingenious solution for the space problem that a PLCC solution would have.

kipper2k, 2021-02-20:
@c0pperdragon   Thanks for the comment, I have just designed an a500/a1k/a2k design that should fit in all of these models. Hopefully it is all good and if proven good i will release the gerbers on here. I have a couple of other ideas as well that if they pan out i will share.

culexus8, 2021-02-23:
Ohh sweet😁 this I will order and build😁 thank you for this nice design.

tonnyrh, 2021-02-23:
Perfect, just perfect ;-)
Should go nicely with my two Vampire's...

culexus8, 2021-02-24:
This can also be used on the A500 if we make a new adapter for Denise so it only use the ffc cable over to your board klipper2k. This may give more room for other addon boards, I see allot of People strugle with space in the A500.

tonnyrh, 2021-02-24:
1. I uploaded the Gerber file to PCBWay, and they found an issue.
"For the PCB order #W373462ASR4_A600 Denise pushon.zip,here's an issue:
here's the rectangle silkscreen in every single layer in the Gerber,here must be something wrong when you export the file?
Please revise the file and upload correct one for review."

**I might dive into it, but just a note.**

2. Any thoughts on how to put OSD Gotek connected?

[image]

kipper2k, 2021-02-24:
i had this done by jlcpcb so not sure what issue is with pcbway. The inner square is identified by dimension layer

seb132, 2021-02-24:
> 
> 
> Wow! This is an ingenious solution for the space problem that a PLCC solution would have.

kipper2k is always thinking out-of-the-box to resolve problems with simple solutions... 👍

culexus8, 2021-02-24:
I did my order with kippers2k gerb files on jlcpcb and had no problems. I was going to order it from pcbway but just never tested jlcpcb before.

kikendo, 2021-02-24:
This does not look like it will fit if you have an ACA620, ACA630 or Fúria.

kikendo, 2021-02-24:
This does not look like it will fit if you have an accelerator.

kipper2k, 2021-02-24:
it will fit, it sits lower than all of the custom chips
all a600 accels will fit

culexus8, 2021-02-24:
looks like it will fit yes.

kipper2k, 2021-02-24:
just to prove it, here is one of my a600s...
[image]

kipper2k, 2021-02-24:
k, 1 more picture to prove its not an illusion...

[image]

tonnyrh, 2021-02-25:
PCBway finally accepted, but:
jlcpcb  charges 6 $ for all (5 of each PCB=10), while PCBway charge 56 $ !!!
...Thats a nobraner where my order goes ;-)
[image]
[image]

kipper2k, 2021-02-25:
yah, pcbway charge more for any design that uses 0.005" traces, if i had used 0.006" then price would be comparable. I prefer to use the smaller traces for boards that need  a high density of traces so in this case JLC wins.

ryanm101, 2021-03-01:
I was trying to do something like the QSB here: https://castlemaniagames.com/products/xstation

struggling with kicad to get the traces and the breakout points right.. but the idea was to mimic the install points against the resistor..

My other plan B was if i just created a board that fitted in the RF slot and mounted to the RF PCB holes I could pull the 5v and GND from the RF pins then just run the short wires from the resistor points to there and then mount the Pi to the PCB that is over the RF area. Further thought was was their a way to break the HDMI port to that board and connect the Pi HDMI to the board instead.. but i think angles and size of solder points might be too difficult.

abrugsch, 2021-03-04:
@kipper2k what's the value of C5? same as C4 or higher?

culexus8, 2021-03-05:
I will use the same as C4, it is just the 5V power side so if there is any problem you can always add more later.

kipper2k, 2021-03-06:
yup, C4 is 0.1Uf

culexus8, 2021-03-06:
Did you test any shorter ffc cable to see if you got rid of any artifacts?

kipper2k, 2021-03-06:
I am waiting for the 6cm cables to arrive from ebay. Unfortunately there is no local purchase option

culexus8, 2021-03-07:
Aha, I will get some 10cm soon I hope, you know ebay is slow😊 still have some parts to order from mouser, will see if the may have some in stock just to test with.

abrugsch, 2021-03-10:
I have 11cm coming from LCSC. parts are in the country. boards... still in china 😭

kipper2k, 2021-03-12:
here is a little video of the hdmi adapter using a shorter cable i got from digikey. 

here is link... http://www.kipper2k.com/pics/a600hdmi.mp4

abrugsch, 2021-03-12:
How long is that @kipper2k ? My 11cm ones arrived today, but boards haven't left china yet... 😭

kipper2k, 2021-03-12:
link is 5 posts up :)

abrugsch, 2021-03-12:
Bleh... I somehow read your last post as ebay so thought it was different to your digikey link... Well friday evening is already in full swing so i blame the Guiness 🤣

culexus8, 2021-03-12:
When I did the first A600 mod I also had some problems, but it all was how I placed the ribbon cable inside the A600 and also changing the powersupply (ordered a new one from icomp). But keeping the cables short is a good idea, so hope the ones from mouser that is 75mm long will performe. Just waiting on parts from mouser to get this up and running.

tonnyrh, 2021-04-01:
U have soldered up and made mine work. Basically used same components as on my Amiga 500 version that works excellent. I do not get white dots but a bit «jaggered» screen.

kipper2k, 2021-04-01:
some hdmi cables are the culprit and not just the device itself. I have seen cheap ones work while expensive ones fail.  I will redo design a bit to use surface mount 16 pin IDE type cable, i have a few things on my plate atm so it would be a couple of weeks.. A shorter cable does make a difference.

tonnyrh, 2021-04-01:
> some hdmi cables are the culprit and not just the device itself. I have seen cheap ones work while expensive ones fail. I will redo design a bit to use surface mount 16 pin IDE type cable, i have a few things on my plate atm so it would be a couple of weeks.. A shorter cable does make a difference.

I use the same cable and setup as for my 500 as shown here: https://youtu.be/G7NAuNks2u4

But this is (great) engineering in progress and a hobby, so its what it is ;-)

c0pperdragon, 2021-04-02:
You can tweak the csync sampling in the settings menu.  It ist the item with values like +ve -ve  or so.

tonnyrh, 2021-04-02:
3'rd try.

Seems like my problems is connected to the blue color.

Question is if its the Raspberry, the RGB2HDMI mainboard, the ribbon or the IC adaptor....

https://youtu.be/cyXWDphq5f4

culexus8, 2021-04-06:
Hi :) Finaly got the time to test this design out and as you may expect I also got trouble whit this adaptor, testet 3 different FCC cables from 15,10 and 7 cm long. All of them gave me fussy image and problems. So I then wraped the cable in copper foil to make a shield and also grounded it to see if there was any changes. But no improvement where made, so I digget out my scope to see if I would find any problems with the clock signal, and yes it was not like it should be. The timing was off compared to how it should be, I then supected that there might be the clock signal that was wrong. Since I did the  A600 board that c0pperdragon designed "small" version I discovvered that if I used the 7Mhz_A signal I got perfect timing compared to R3 color. So I then looked at the kipper2k breakout board and found the issue right away, the cdac was connected to pin37 of denise. so by changing this to pin38 and feeding the 7Mhz_A signal to the cdac point of the breakout board gave me rock solid picture even with 15cm FCC cable.  So if some one else would try this out and confirm that this is the problem, please do so :) No tweaking where done in the software, all run perfect with stock settings so far. 

[image]

tonnyrh, 2021-04-06:
Confirming. Works great now ;-)

Thank you all ;-)

https://youtu.be/8wrpzLcjNGs

culexus8, 2021-04-06:
Hey Kipper2k:) 

I think your board is just a perfect board for all of the systems, your board is versatile and can be used for many more systems than just the A600. What if there was made a breakout board that you could solder on the underside of the A500 motherboard  so the denise just stays put and wee feed the signals trough the FCC cable to you board. I think there should be space for it. 

The big thing about it is that it will give space for any accelerator that sists in the cpu socket. With the current design for A500 it gets tight. It is a good desing but we need more space  :)

kipper2k, 2021-04-07:
Thanks for the kind words, My CDTV version will be here this week, it will allow the closing of the case and has a nice plate with it. My new a500 version allows the cable to sit towards the rear of the case and will fit in the A1000's also. I'll release the new a500 version soon and will test the CDTV version out. The new A500 design will allow the upcoming "buffee" and the new multi function A500 addon board i have almost finished to fit inside the a500 case with the case closed also :)

solarmon, 2021-04-07:
I thought that the 7Mhz signal should be used for an ECS Denise? And that CDAC should be used for OCS Denise. That is what the jumper is for on the RGBtoHDMI v2 board.

culexus8, 2021-04-07:
> Thanks for the kind words, My CDTV version will be here this week, it will allow the closing of the case and has a nice plate with it. My new a500 version allows the cable to sit towards the rear of [...]

Sounds nice :) looking forward to see the new boards. Do you have any more info on your multi function A500 addon board and what is ?

culexus8, 2021-04-07:
> I thought that the 7Mhz signal should be used for an ECS Denise? And that CDAC should be used for OCS Denise. That is what the jumper is for on the RGBtoHDMI v2 board.

I think you are totaly right. :)

kipper2k, 2021-04-08:
i have redone the pushon board, here is updated pic of it...
[image]

ad here is the updated gerber file, it is now V1.0 which is still compatible with my original Denise board

[A600 Denise pushonv1.zip](https://github.com/c0pperdragon/Amiga-Digital-Video/files/6278541/A600.Denise.pushonv1.zip)

kipper2k, 2021-04-09:
k, i redone the layout for the a600 HDMI2RGB board and this is what i came up with. It should not interfere with any addons. I will release the STL file for the support once i finished tweaking.

[image]

[image]

[image]

[image]

[image]

[image]

 Picture quality is great,  :)

abrugsch, 2021-04-09:
That looks very neat @kipper2k. Thank you for the updated file. I've ordered some from JLC to go with my initial batch. (If only I'd remembered to use my "jlcfactory" code but meh. It's only 2 bucks off...)

abrugsch, 2021-09-15:
@kipper2k did you ever finish tweaking the STL file for the support?

Aether2013, 2021-11-20:
Does anyone have a BOM for the pimate2?

culexus8, 2021-11-20:
> Does anyone have a BOM for the pimate2?

What is a pimate2?

Aether2013, 2021-11-20:
The board that connects the A600 Denise Pushon, to the Pi-Zero.

[image]

GiuSimoo89, 2025-12-15:
https://amigastore.eu/974-rgb2hdmi-amiga-600-ffc.html
ho acquistato questo ma fa schifo lo zoccolo PLCC e vorrei mettere il tuo "A600. Denise.pushonv1". questo è l'ultimo vero?
si potrebbe avere già tagliato al centro? è compatibile con il tuo pcb? ora vado a controllare, ma intanto chiedo. grazie.