c0pperdragon/Amiga-Digital-Video #39: Pin header strip alignment

solarmon, 2021-03-05:
Hi,
I just built my first RGBtoHDMI v2 board but having some trouble with the pin header strip alignment to the Denise socket on the motherboard (of an rev 8A A500plus).

Using a bare board you can see how the holes line up with the Denise socket holes.

[image]

I'm using the pin headers from:

https://lcsc.com/product-detail/Pin-Header-Female-Header_BOOMELE-Boom-Precision-Elec-C9742_C9742.html?fbclid=IwAR2bzglgs2G66_9C56nVjPR7m6qyY_YiOLFeSNOrObG6v17rgG3hsfyci2M

Which are different from the longer and thinner ones like at:

https://www.ebay.co.uk/itm/Uk-2-Gold-Plated-2-54mm-40-Pin-Single-Row-Straight-Round-Pin-Header-Strips/324430051449?hash=item4b8989b879:g:hWEAAOSwglZf6ybg

When I use the pin headers from LCSC, they do not seem to be line up with the Denise socket holes and I cannot get it to fit in to the socket:

[image]

Can somebody offer some advice on what header pins should be used and if there is any special build instructions for them.

Thank you.

kipper2k, 2021-03-06:
for the header pins to connect to Denise socket you should use something like these...

 search in ebay "5Pcs Gold Plated 2.54mm Male 40 Pin Single Row Straight Round Pin Header Strip". Look for the best prices.  I set the vias at 0.025" drill and 0.050" diameter, this allows traces to be fed through the pind with lots of room

solarmon, 2021-03-06:
@kipper2k 

Yes, I found those types of headers, as per my eBay link. I actually found some I have purchased previously, so I will try a build with those.

The short thicker round pin headers just don't have enough give, but it may still be possible if it was angled slightly during the build.

solarmon, 2021-03-06:
I tried with the longer/thinner pin headers and they definitely fit better!

[image]

Oh well, lesson learnt. Those other pins are just too thick and doesn't have much give.

kipper2k, 2021-03-08:
The  longer thiner ones are better as they dont stretch the sockets too bad and for the end user who puts the socket in, it is less likely to be removed too often and they are easier to replace if broken