c0pperdragon/Amiga-Digital-Video #40: Pi Zero relocation adapter boards - video artefacts - pixel 'sparkles' on edges

solarmon, 2021-03-05:
Hi,

I'm testing out several Pi Zero relocation adapters - there are three types: A, B and C:

Type A is just flips it over:

[image]

Type B shifts it sideways and flips it:

[image]

Type C shifts it sideways, flips it and rotate it:

[image]

Type A works fine without any issues.

However, type B and C has video artefacts where on certain edges there are pixel 'sparkles' artefacts.

I'm aware that this is due to the increased signal traces, but is there anything that could be done to improve it?

Thank you.

kipper2k, 2021-03-06:
shorter tracks definitely increase performance, if you increase track lengths then you should ensure that decoupling caps are as close to IC's as possible. Also extending the board over other areas of the Amiga motherboard can also pickup unwanted noise from the motherboard so you may have to reduce board length and location to reduce the noise

solarmon, 2021-03-06:
@kipper2k 

Regarding your comment about caps being close to IC's as possible. These are just relocation boards for the Pi connector. I'm not changing anything with the RGBtoHDMI boards.

The artefacts only appear at 'edges' - where an area of colour changes to another. It's hard for me to capture that, but you can see the white pixel in this (poor quality) picture. These white pixels appear and disappear randomly on such 'edges':

[image]

Why would it only appear at these edges?

solarmon, 2021-03-06:
I'm actually seeing this shimmering/sparkle artefact using a normal RGBtoHDMI setup - i.e. no relocator or anything.

With the normal setup setup, in the Amiga Test Kit screen, I don't see these artefacts.

However, when I load up workbench, the artefacst appear on the right hand side of any black areas next to lighter areas.

For example:

[image]

It is always on the right hand side of the contrasting edge, if that is of significance.

I haven't checked the RGBtoHDMI menu options yet. Maybe there are some settings I need to tweak?

de-nugan, 2021-03-06:
This is the same fault I wrote about here =>  https://github.com/c0pperdragon/Amiga-Digital-Video/issues/11#issuecomment-769056224

I'm also using a slightly longer connection through a very short IDC cable. I tested a longer cable and saw increased sparkles and artifacts so the length does appear to be critical. I haven't been able to test the 'default' setup as I don't have any of the female pin sockets, so I may have to solder the Pi to the board and try that instead.

solarmon, 2021-03-06:
So I think I have narrowed down the issue a bit. It seems this issue is only specific to this X-Copy disk I'm using to test. Everything else I've tried so far has been rock solid. The X-Copy disk I'm using is created from the xcopy_1992-11_en_1200.adf image (http://jope.fi/xcopy/zip/xcopy_1992-11_en_1200.zip). If somebody is able to test this out and check if they have the same issue. (You need to press F10 to go in to Workbench mode).

solarmon, 2021-03-06:
> This is the same fault I wrote about here => [#11 (comment)](https://github.com/c0pperdragon/Amiga-Digital-Video/issues/11#issuecomment-769056224)
> 
> I'm also using a slightly longer connection through a very short IDC cable. I tested a longer cable and saw increased sparkles and artifacts so the length does appear to be critical. I haven't been able to shorten the path as I don't have any of the female pin sockets, so I may have to solder the Pi to the board and try that instead.
> 
> My guess is that the increased path length is making it harder for the flipflops to drive the lines high from 0v (black) so we are losing bits resulting in colour sparkles where there should be grey or mid-tone colours. I aint no engineer though..
> 
> Oh and sorry to disappoint you, I've tried just about everything in the menus. I don't think it can be fixed there, likely only in hardware or perhaps firmware.

If this issue with my relocator boards can't be fixed then I still have the option using my custom RGBtoHDMI that I'm designing and which relocates the Pi board too. Hopefully that won't have the same issue, but  I still need to get these custom boards made and tested.

IanSB, 2021-03-07:
@solarmon 
Try replacing the 74LVC parts with 74VHC parts. These are slower and fixed the sparkly problem on my rev5 which doesn't have the clock filtering components that are fitted on rev6 & later boards which slightly delay the clock signals.
Note they must be 74VHC parts which are 5v tolerant, not 74HC parts which are not 5v tolerant when powered at 3.3v

@de-nugan 
I also suggest you try the above to fix your issue in  https://github.com/c0pperdragon/Amiga-Digital-Video/issues/11

solarmon, 2021-03-07:
@IanSB 

Thanks for the suggestion for replacement parts. When you say 'these are slower' - do you mean the 74LVC  or the 74VHC part?

I had already got a batch of the 74LVC parts because that was what was suggested in the schematics. ='(

For reference, these are the parts from LCSC:

U1!: https://lcsc.com/product-detail/74-Series_Nexperia-74LVC86APW-118_C6102.html
U2 and U3: https://lcsc.com/product-detail/74-Series_Nexperia-74LVC574APW_C6098.html

Can you confirm these are the suitable replacement.

U1: https://lcsc.com/product-detail/Pre-ordered-Products_ON-Semiconductor-ON-74VHC86MTCX_C903423.html
U2 and U3: https://lcsc.com/product-detail/Flip-Flops_ON-Semiconductor-ON-MC74VHC574DTR2G_C364877.html

However, the U1 74VHC86MTCX part is not in stock on LCSC.

Are you suggesting that U1, U2 and U3 should all use replacement parts?

IanSB, 2021-03-07:
@solarmon 
>When you say 'these are slower' - do you mean the 74LVC or the 74VHC part?

The VHC parts are slower. Their propagation delay is about double that of the LVC parts and that allows a longer settle time before latching the data bits.

>Are you suggesting that U1, U2 and U3 should all use replacement parts?

Yes although changing U1 (74LVC86) which handles the clock signal has the most effect so try that first if you are changing parts on an existing board.

>Can you confirm these are the suitable replacement.

They look to be the correct family although I used these toshiba parts from RS:
https://uk.rs-online.com/web/p/products/1713480/
https://uk.rs-online.com/web/p/products/1713428/

solarmon, 2021-03-07:
@IanSB 

Thanks for the parts link from RS. I'm UK based, so that suits me fine.

BTW - these issues (sparkles due to relocation board, and sparkles using a specific version of X-Copy, and Xenon II) happens on a n A500 rev 6A and an A500plus rev 8A.

IanSB, 2021-03-07:
>Thanks for the parts link from RS. I'm UK based, so that suits me fine.

The VHC86 parts have gone up a lot in price, I paid £0.056 each a few weeks ago, they are now £0.217 each.
The VHC574 parts haven't gone up (yet!)

>BTW - these issues (sparkles due to relocation board, and sparkles using a specific version of X-Copy, and Xenon II) happens on a n A500 rev 6A and an A500plus rev 8A.

Can't guarantee they will fix your issue but they did fix mine.

de-nugan, 2021-03-07:
To achieve the shortest possible data path I've just soldered the Pi to the adapter board and this appears to have fixed the sparkle problem. In WB3.1 the sparkles and glitches are now gone, and the screen geometry is now absolutely bang-on pixel perfect straight out of the box. 

This is on my Rev 5 (Assy 312513) A500. The parts used were Nexperia 74LVC574APW.

I'll be building more boards for some other Amigas, do you know if the VHC tip applies to any of the A/B2000 revisions?

matthiesenj, 2021-03-08:
For HC parts, you should use VHCT or AHCT, which have TTL-compatible inputs.
The point about propagation delay is probably relevant, since the original c0pperdragon design relies on the propagation delay of two gates to generate the 14MHz flipflop clock.

solarmon, 2021-03-08:
@matthiesenj 

I'm confused now - @IanSB mentioned '74VHC' parts and now you are mentioning 'VHCT' parts! Are they different? I've already ordered the  '74VHC' parts.

matthiesenj, 2021-03-08:
@solarmon your Amiga uses 5V TTL signal levels - the inputs on the (T) variants are specifically designed for the TTL low/high signal thresholds. If you already ordered parts, you can just try them - IanSB says they work for him and they won't hurt anything, but since this thread is about signal integrity problems, I thought I would mention it :-) The AHCT also has schmitt-triggers on the inputs which can also help correct sampling of noisy input signals.

If you're exchanging parts on a populated board, try changing the 86 first to see if it fixes your glitches..

solarmon, 2021-03-08:
@matthiesenj Thanks for clearing that up!

It's going to be easier for me to build new boards - re-working the existing ones probably not going to work with just a normal soldering iron!

solarmon, 2021-03-09:
@c0pperdragon 

I'd thought I'd reply in this ticket, as opposed to https://github.com/c0pperdragon/Amiga-Digital-Video/issues/11#issuecomment-793126725

This is my setup, including my relocator. You can also see how I have made my bank of 27pF capacitors:

[image]

Connecting this bank of capacitors to PiSYNC (GPIO23) and GND does not seem to make any difference to the sparkles.

However, if I connect it to PiCLK (GPIO17) I do see a significant difference - it doesn't get rid of it completely, but there is a visible difference. I'll try adding more caps to the bank and see if it makes any further improvements.

solarmon, 2021-03-09:
@c0pperdragon LOL! It happens! No problem.

What I'm also finding is that different RGBtoHDMI v2 boards have slightly different behaviour. The difference at the moment, is that  the one with the thicker pins has less artefacts than the thinner pins. I'm not sure whether it is because of those pins, or something slightly different in my soldering of those boards. It's only small sample at the moment, but there is some difference.

How this effect what capacitance value to use for PiCLK I need to try to test. Also, when I get the '74VHC' parts I will find out whether that makes a difference too.

solarmon, 2021-03-09:
@IanSB 

I meant the different RGBtoHDMI v2 boards I currently have, all using the same chips. The only difference is that the first one I used the thicker pins. Subsequent ones I used the thinner pins.  The first one (with thicker pins) seem to have less video artefacts.

But I can't come to any explanation yet since the this is only testing with a few boards.

IanSB, 2021-03-09:
>I meant the different RGBtoHDMI v2 boards I currently have, all using the same chips. 

I know, what I meant was the difference in noise is due to the chips themselves. If you swappped the chips on two of your boards the noise would swap as well.

solarmon, 2021-03-09:
> I know, what I meant was the difference in noise is due to the chips themselves. If you swapped the chips on two of your boards the noise would swap as well.

Ah, right, gotcha! That is very surprising then, that it would cause that much of a difference.

c0pperdragon, 2021-03-09:
The whole matter of the parkly pixels is very confused by the fact that there are two completely independent possible causes:

1. The adapter can not reliably latch the color data. This can be caused by using the wrong data sample point (Denise vs. SuperDenise jumper) or by some other glitchy behaviour of the main board. Specifically A500 Rev.5 boards seem to not filter the CDAC clock signal properly. I guess this is mainly causing IanSB's trouble. One cure seems to be to use logic ICs that are less sensitive to high-frequency noise. 

2. The signals from the adapter to the Pi do not arrive at a suitable time. The adapter adds only a very short extra delay (2 LVC gate transition times) to the output clock. So the Pi seems to not always be able already see the correct color data on the edge transitions of the clock. This is especially true when using longer leads between adapter and Pi. I proposed to just add a small capacitor to delay the clock a bit. This seems to work in some circumstances. A more proper soution would be to add more logic gates in the signal path and use their transition times. Maybe a series of simple inverter gates could do the trick.

matthiesenj, 2021-03-09:
@c0pperdragon very true about the 2 possible causes.

Regarding 1), did you measure the (hires) pixel transition timing in relation to e.g. /CDAC for denise/superdenise? Are they borderline with respect to the flipflop setup/hold times?

c0pperdragon, 2021-03-09:
I did measure the Denise timings and the were pretty safe when using the edges from _CDAC.
Other users did try SuperDenise with _CDAC edges amd there sampling was very borderline. Sometimes it worked just fine, sometimes it caused glitches. So I extrapolated that the sweet spot for SuperDenise would be right between the edges of _CDAC and therefore used the edges of the 7MHz clock instead. This seemed to work basically flawless on SuperDenise so far.

solarmon, 2021-03-09:
@c0pperdragon The 74VHC chips came and I built another RGBtoHDMI v2 using these chips.

I can happily report For a normal setup (no relocator) I now do not get any percieveable sparkles when using X-Copy (Oct 92) workbench or when using DiagROM 1.2.1 (I recently noticed the issue with this recently, but hadn't reported it yet), or in Xenon II (1 disk version).

Even using my relocator board, this has drastically reduced the sparkles to nearly gone. I can still see pink sparkles (on blabk/light grey edges) if I look very closely/carefully. Maybe that could be improved further with a suitable cap across PiCLK and GND, but I'm very happy with the results!

IanSB, 2021-03-09:
@solarmon 
>I can happily report For a normal setup (no relocator) I now do not get any percieveable sparkles when using X-Copy (Oct 92) 

Thanks for confirming they fix your issues. I suspect the problems with the extender boards are caused by reflections due to the long PCB tracks.

FYI I have designed an Amiga pickup board for the CPLD version which allows the Pi to be placed anywhere on a long ribbon cable (mainly meant for external use) which is similar to the Atari solution shown in https://github.com/c0pperdragon/Amiga-Digital-Video/issues/6#issuecomment-770140199

LinuxJedi, 2021-03-09:
> Thanks for confirming they fix your issues. I suspect the problems with the extender boards are caused by reflections due to the long PCB tracks.

In which case a resistor may help with the reflection?

matthiesenj, 2021-03-09:
@LinuxJedi yes, series resistors on the flipflop outputs could help eliminate reflections - some logic families even incorporate series output resistors on-chip, but I haven't found such a device to directly replace the LVC574..

https://www.ti.com/lit/an/scba012a/scba012a.pdf

LinuxJedi, 2021-03-09:
> @LinuxJedi yes, series resistors on the flipflop outputs could help eliminate reflections - some logic families even incorporate series output resistors on-chip, e.g. AHC574 (which can directly replace the LVC574)..
> 
> https://www.ti.com/lit/an/scba012a/scba012a.pdf

Are you sure about the AHC574? I can't see it on the datasheet. I believe the NXP ICs have it in LVC and similar if they have a '2' in them, but there is only the 374 in that so it would need rewiring.

solarmon, 2021-03-10:
@c0pperdragon 
Is this what you were trying to do with this part of the circuit?

[image]

I wouldn't have the expertise to try to find a suitable fix, but what about feeding back the PiCLK signal to one of the spare not used input/output on the existing logic chip? Or will that still not delay it enough?

LinuxJedi, 2021-03-10:
I think whilst @c0pperdragon's v1/v2 solution is fantastic work and for a majority of people it will work great, @IanSB may be right and a xc9572xl based solution may be the way to solve these minor issues long-term. Should even be able to fit it into the same physical profile as the c0pperdragon adaptor.

c0pperdragon, 2021-03-10:
I still think this is total overkill for an internal solution. It is probalbly really only a matter of fixing the setup times.

solarmon, 2021-03-10:
> Yes, I used the one spare gate on the XOR for delay. It seems that this is yet not enough in sone circumstances. The free bits on the flipflops can not be used for that, though.
> When you are using the adapter jumpered to Denise mode, you could actually repurpose one of the XOR gates. But this requires modifying the adapter in some irreversible way. But maybe this one gate transition would be just enough.

I'm designing building my own RGBtoHDMI board, that also relocates the Pi. I could add this to the design (as well as an optional cap on PiCLK) and try it out.

solarmon, 2021-03-10:
@c0pperdragon 

Oh, you mean the U1 chip. I was talking about the spare input/output on the U3 chip, Would that work - using U3?

solarmon, 2021-03-10:
@c0pperdragon 

OK, understood.

Would a dual inverter like these 74HC2G14-Q100/74HCT2G14-Q100 be suitable.

[image]

I was looking to remove the 3.3V regulator anyways, so this could just fit in its place footprint wise.

LinuxJedi, 2021-03-10:
Really dumb question time based on me looking at the Denise datasheet: would an AND of 7MHz and CDAC to the flip-flops clock work with straight CDAC to the PiCLK? Or is the flip-flop latency also part of the problem?

solarmon, 2021-03-10:
Actually, this SN74LVC2G04 seems more suitable (or equivalent), as the output should be 3.3.V instead of 5V.

[image]

c0pperdragon, 2021-03-10:
The HC inverters are probably better. HC is slower than LVC which is exactly what you need for delay. Also the signals edges are not so sharp and will cause less trouble.  HC can also run with 3.3V without issue  (not HCT, though!).  
Schmitt-Trigger characteristic (74HC14) is not necessary, but should not hurt either.

c0pperdragon, 2021-03-10:
> Really dumb question time based on me looking at the Denise datasheet: would an AND of 7MHz and CDAC to the flip-flops clock work with straight CDAC to the PiCLK? Or is the flip-flop latency also part of the problem?

The flip-flops sample the bits at the positive edge of the clock input. So you need something that goes high twice in a single system clock cycle (high-res pixels are generated at about 14MHz). This is done with by strangle looking XOR circuit which goes high whenever the input clock changes.
AND-Combining CDAC and 7MHz would only generate a 7Mhz clock with 25% duty cycle. I don't know what this should achieve.

solarmon, 2021-03-10:
@c0pperdragon 

I came up with this updated schematic to add in the dual inverters and the option switch. Is this what you had envisaged?:

[image]

LinuxJedi, 2021-03-10:
> The flip-flops sample the bits at the positive edge of the clock input. So you need something that goes high twice in a single system clock cycle (high-res pixels are generated at about 14MHz). This [...]

Ah! The bit I was missing was the high-res 14MHz. I was thinking that this would help with the time between 7MHz rising edge and making sure that the colour lines were setup. IIRC 7MHz rises first and then CDAC so reading between the rise of CDAC and fall of 7MHz seemed like something that might work. But yea, high-res screws all that up :

matthiesenj, 2021-03-10:
@LinuxJedi yes, according to the document I linked, TI AHC parts are "light drive".. In the 74AHC574 datasheet it is also mentioned that it's designed for driving capacitive/low impedance loads and only has 4mA drive capability: https://www.ti.com/lit/ds/symlink/sn74ahc574.pdf

c0pperdragon, 2021-03-10:
@solarmon
Yes, this is what I meant.  If you leave away the voltage regulator and draw the 3.3V from the Pi, you should have enough space for that.

solarmon, 2021-03-10:
@c0pperdragon 

I still need to figure out and finalise the trace routing (it's a bit of a squeeze!) but I was hoping the board layout would be something like this.

[image]

[image]

c0pperdragon, 2021-03-10:
This looks actually pretty neat. It is hard to follow the details, specifically how the power supply is connected, but I just trust you on that.

solarmon, 2021-03-10:
@c0pperdragon Had to redo some of the traces to fit everything in, but I've managed to connect everything up.  I've sent you the KiCAD files for you to take a look at.

[image]

matthiesenj, 2021-03-10:
@solarmon did you exchange the 1uF cap for a 10nF one? I don't think that's a good idea - you should rather increase it if anything.

solarmon, 2021-03-10:
> @solarmon did you exchange the 1uF cap for a 10nF one? I don't think that's a good idea - you should rather increase it if anything.

The 10uF cap is for the 3.3V supply from the Pi. I took the power supply design from the RGBtoHDMI small adapter board design.

solarmon, 2021-03-10:
Oops! Looks like C5 is meant to be 10uF, not 10nF.

solarmon, 2021-03-10:
I thought I'd also add FlashFloppy OSD support, as per https://github.com/c0pperdragon/Amiga-Digital-Video/issues/26

The BP1 connection goes to PB15 on the Blue Pill and the R2 1K resistor goes to Pin 18 (GPIO24) of the Pi header.

[image]

IanSB, 2021-03-11:
@solarmon 
>I came up with this updated schematic to add in the dual inverters and the option switch. Is this what you had envisaged?

That may not work as you are only delaying the PiClk signal into the Pi and I think the problem is that the 574 chips have already latched invalid data. From my own tests the only thing that worked at all was delaying _CDAC before feeding into the rest of the system.

c0pperdragon, 2021-03-11:
@IanSB  This heavily depends on the actual cause of the issue, as there are two completely distinct possibilities.

matthiesenj, 2021-03-11:
How does the Pi read the pixel data? Are all the GPIO inputs continuously polled as one 32 bit word, then monitored for changes to the PiClk bit, and if changed, the rest of the word fed on for further processing? Or does it read the GPIO port multiple times per pixel?

solarmon, 2021-03-11:
I was testing my VHC parts build and found a colour combination that caused a different kind of sparkling artefact. These were sub-pixel (Amiga pixels) sparkles, which are different to the full-pixel (Amiga pixels) sparkles I saw previously.

[image]
[image]
[image]

IanSB, 2021-03-11:
@solarmon 
>These were sub-pixel (Amiga pixels) sparkles,

What resolution were you using at the time?

solarmon, 2021-03-11:
> What resolution were you using at the time?

@IanSB 

It was just OCS standard 320×256 LowRes 4 colours.

When I say "sub-pixel" I meant that the sparkles artefacts were much smaller than the Amiga pixels itself, as can be seen by the pictures. As if it is something to do with the HDMI graphics processing of the monitor at 1080p resolution that was somehow affected. 

I thought it was a problem with my HDMI cable setup - I was using a 1m flex cable at the time. Powered off and swapped  to a normal HDMI cable and I could not replicate the issue. However,, when I put back the HDMI flex cable, I still could not replicate the issue again. and I still can't, so I can't troubleshoot it further.

IanSB, 2021-03-11:
@ solarmon
>As if it is something to do with the HDMI graphics processing of the monitor at 1080p resolution

That definitely looks like HDMI noise as the pixels are too small to be a capture side problem. Perhaps it was caused by a HDMI cabling issue due to poor connection on one of the signals which went away when everything was reseated. Are you using a mini HDMI to HDMI adapter with the cable as that can exacerbate things like that.

You can test for HDMI problems by using the screencap option (short press of button). If the HDMI output is noisy but the screencap png files on the SD card are clean then the problem is on the HDMI interface and likely a cabling issue. (There is an option that can be added to Pi's config.txt file (config_hdmi_boost) to increase the switching strength of the HDMI signals and that can help with marginal cabling problems)

solarmon, 2021-03-11:
@IanSB 

Well, the issue was similar in the sense that as I moved the colour slider, the artefact 'noise' changed - either increase, reduced or disappear based on the colour.

Until I get this issue again then I can't troubleshoot it further I know next time I will reseat the HDMI connector first, or swap HDMI cables, without powering down.

solarmon, 2021-03-11:
> You can test for HDMI problems by using the screencap option (short press of button). 

Thanks! I was not aware of this feature - it will come in very useful!

solarmon, 2021-03-12:
@c0pperdragon 

Would a DS1100L be even more suitable to add the delay? You can select delays between 4ns to 500ns.

[image]

[16 more replies]