c0pperdragon/Amiga-Digital-Video #42: Strange video issue

aiobofh, 2021-03-11:
Hi there!

I have this very strange video issue. Picture looks crisp clear, but with three ribbons of weirdly (wedge-)shaped garble horizontally..

When first turned on, they move a (very) little bit vertically, but after that, they stay where they are on the photo.

Any spontaneous ideas?
[image]

aiobofh, 2021-03-12:
... Addition.... I tried another monitor than the TV i took the photo of above.. (That TV has been quite friendly with all other sources i've thrown at it... So this is weird.). Result on my workstation monitor: 

[image]

... How come?

IanSB, 2021-03-12:
@aiobofh 
>How come?

The first monitor probably doesn't like the non-standard timings produced by the genlock.

First thing to try is manually changing the resolution to 1920x1080 and the refresh to 50Hz. (Assuming that's a 1080p monitor)
This will produce a more standard 1080p 50Hz output but it will still be a little off frequency due to the genlock.

If that doesn't work then try switching the genlock off in the settings menu although that means the video output won't be locked to the input so you might get tears and dropped/repeated frames.

aiobofh, 2021-03-12:
@IanSB right you are!

Strange that that otherwise friendly TV did not like it. Anyhow. End-goal was to have crisp up-scaled picture in the livingroom TV. Just tried it out there and it works fine! Thanks for the tip!

End result:

[image]

closed 2021-03-12 by aiobofh