c0pperdragon/Amiga-Digital-Video #44: Scanline Level

ransom1122, 2021-03-18:
Can anyone tell me if scanline level option in the OSD works? I can't seem to change the value from 0?

Enabling scanlines work, but not scanline level?

Thanks

IanSB, 2021-03-18:
@ransom1122 

Variable intensity scanlines require additional processing and to get them working means significantly overclocking the Pi zero so they are disabled by default.
However there is an additional profile called "Amiga_Var-Scanlines" which has the multiple changes to settings required for variable scanlines so change your profile to that option and you should then be able to set the scanline level.
Note that scanlines are disabled when the menu is on screen and your Pi may hangup due to the overclocking in which case I suggest adding a heatsink.

ransom1122, 2021-03-18:
Looks like I'll pass, I dont have a heatsink and not interested in overstocking. The standard scanlines are good enough as it is.

Jimiga, 2021-03-18:
Thanks for the reply @IanSB . I'm interested in this too, just wondering - with a heatsink installed, is there any risk to damaging the pi with the overclocking involved here? 

I think standard scanlines will be fine, but I'm looking forward to experimenting when my setup arrives next month, and I'm learning what I can in the mean time. Thanks!

IanSB, 2021-03-18:
@ransom1122 
>Out of interest for others that are wanting more, where does one find that amiga var scanlines profile

In the main menu, change the "Profile" setting.
Options are "Amiga", "Amiga Blk-Scanlines", "Amiga Var-Scanlines"

@Jimiga
>with a heatsink installed, is there any risk to damaging the pi with the overclocking involved here?

I don't think so, it usually works with no heatsink anyway so not really stressing the chip that much.

See this thread for some example screencaps of variable scanlines:
https://github.com/c0pperdragon/Amiga-Digital-Video/issues/16

abrugsch, 2021-03-18:
> Looks like I'll pass, I dont have a heatsink and not interested in overstocking. The standard scanlines are good enough as it is.
@ransom1122 try combining the scanlines with softening the picture (one of the menu options) i found the combination gives a very pleasing near-CRT effect. Though your mileage may vary so give it a try