c0pperdragon/Amiga-Digital-Video #47: non-destructive way of fitting an HDMI output to the Amiga 500.

NielsPlougman, 2021-04-16:
This is a non-destructive way of fitting an HDMI output to the Amiga 500.

When adding the RGBtoHDMI adapter to the original Amiga 500, the problem arises how to get the HDMI cable out of the Amiga.  

I've come up with the following solution:
[image]

This little housing, for a Female HDMI connector, clips right on the Amiga case. I tried to design it in the A520 TV modulator style, so it should not look too much out of place in the 80' computer design.  

[image]

The cables go through the small vents on top of the Amiga case. 
I've made a complete guide on prusaprinters.org, where you can download the STL files and see how to assemble it. 

[https://www.prusaprinters.org/prints/63688-rgbtohdmi-adapter-housing-for-amiga-500](https://www.prusaprinters.org/prints/63688-rgbtohdmi-adapter-housing-for-amiga-500)

Hope someone finds it usefull.

aiobofh, 2021-04-16:
I did like this :) 
[image]

mthiesen, 2021-04-17:
@NielsPlougman This looks very nice, good job! I thought about printing a custom expansion port cover with cut-outs for HDMI and the push button, but this is so much better. Definitely goes onto my ever growning list of things-to-print.

@aiobofh I am curious: what is the story behind those connectors with the LEDs and voltage labels?

LinuxJedi, 2021-04-21:
> @aiobofh I am curious: what is the story behind those connectors with the LEDs and voltage labels?

They will be ATK / DiagROM loopback testers.

krug3rf, 2021-05-04:
My personnal design for single HDMI with or without µSD slot or dual HDMI (PiStorm and RGBtoHDMI), available [here](https://cults3d.com/fr/utilisateurs/krug3r/)

[image]

[image]

[image]

rkeniger, 2023-05-11:
Are you all just ignoring the fact that with this mod in place you can't re-install the RF shielding on the Amiga? The installed Pi Zero totally blocks one of the screw holes.

LinuxJedi, 2023-05-11:
> Are you all just ignoring the fact that with this mod in place you can't re-install the RF shielding on the Amiga? The installed Pi Zero totally blocks one of the screw holes.

Is there a reason you would want to re-install it? I don't know many people who keep it on.

rkeniger, 2023-05-11:
I guess just for completeness? Are there no issues with it off? Since it’s a hefty shield I’m assuming it’s there to prevent the Amiga interfering with radio, but if it’s unlikely to affect things then great - I know pretty much zero about RF when it comes to circuitry.