c0pperdragon/Amiga-Digital-Video #48: Turning off HDMI output programatically

c0pperdragon, 2021-04-23:
I got a request about the possibility to turn off the HDMI output in certain circumstances. The main use would be to allow switching between the HDMI output of a Vampire card and the RGBtoHDMI automatically. But it could also be useful for implementing a simple video power-down mode.

@Ian
I could imagine that the Amiga can be be programmed to turn off/suppress its csync signal (will check this). What would happen then on the side of the RGBtoHDMI? Would it just continue to display the content of its framebuffer, or would it turn off the output alltogether? Could such a behaviour be implemented?

Cjsnet, 2021-04-23:
Thanks for considering this. I am using an auto HDMI switcher which detects a video signal coming from an HDMI input, then switches the output to display that signal. So if the Vampire is transmitting but RGB2HDMI is not, the TV will show the Vampire image. The problem is RGB2HDMI is always transmitting, therefore no auto switching can occur. So this would be a fantastic addition.

Here's a preview of the [thumbnail](https://photos.app.goo.gl/Bu2Y5it2pd41FbNp8) of my video about this switcher technique. If I could announce support for auto switching in the video it would be even better and a great enhancement to the RGB2HDMI! My video won't be finished for about a week. Possibly longer...

Thanks guys.

c0pperdragon, 2021-04-23:
@IanSB  Oops!  I wanted to notify you in the first post, but got our name wrong...

 I did some more research on the matter of turning off the CSYNC from the Amiga programmatically. It may not be as easy, as I had hoped. The registers in the Agnus chip to fine-adjust the sync signal generation were introduced with the ECS chips. Maybe it is also possible by other means, but I am not an expert on the inner workings of the Amiga video hardware.

c0pperdragon, 2021-04-24:
This issue seems to be very complicated and I don't see an easy way to turn off the analog video output.
But the documentation of the picasso96 driver which is also used by the Vampire accelerator hints at the possibility to power down its HDMI output. Maybe this is already good enough for the purpose of auto-switching.

Perifractic, 2021-04-24:
Unfortunately I don’t know anyone on the Vampire team and they haven’t released an update for the 500 in a while. So it may not be likely for the Vampire to provide the feature. Do you think RGB2HDMI can use the same method?

c0pperdragon, 2021-04-24:
The RGBtoHDMI does not have many options here. It only has these few signals to generate the image from. If the Amiga does not change these signals (possibly done by the picasso96 driver) I can see no solution.

Did you check with the documentation for the Vampire and the picasso96 driver if there are some ways to power-down the HDMI output?

Perifractic, 2021-04-24:
I will look into that further. The problem is, it needs to happen automatically, so even if there was a way, I'm not sure how a user would get the Vampire to power down whenever the Amiga switched to a certain graphics mode.

IanSB, 2021-04-25:
@Perifractic 
@c0pperdragon 

I've implemented the basic functionality of this so that if the input syncs disappear then the HDMI output goes into DPMS standby and any connected monitor switches off but if there isn't any way of disabling the syncs on the Amiga then it might not be much help.

Perifractic, 2021-04-25:
Very interesting. How can I test this out? Thanks!

LinuxJedi, 2021-04-25:
> @Perifractic
> @c0pperdragon
> 
> I've implemented the basic functionality of this so that if the input syncs disappear then the HDMI output goes into DPMS standby and any connected monitor switches off but if there isn't any way of disabling the syncs on the Amiga then it might not be much help.

Might be too much work for the loop but could it be possible to add something to detect when all pixels are black for X seconds and DPMS? (revert when there is colour again)

c0pperdragon, 2021-04-25:
I fear, this new feature from the RGBtoHDMI does not really help all by itself. I have tried to play with the hardware registers of the Amiga. I can somehow influence the CSYNC generation in a weird and unexpected way by messing around with the registers of the ECS Agnus. But this all looks pretty broken and there is basically no real documentation on this to be found.

So I think I can not offer you much more help here. Maybe you can set up the Vampire to somehow disable HDMI when the Amiga is not using it. But I can not give you much support there, either.

One thing is interesting, though. You said, when the workbench uses the vampire for output, the analog screen goes gray. But why gray?  Is this something than can be configured somewhere?  Maybe there is some configuration possibility. At least some people would prefer a black screen for this purpose, I could imagine.  Maybe there is also a screen-blanking option, or something.....

Perifractic, 2021-04-25:
Ah well thanks anyway. 

I did find a Vampire setting at the Facebook that will turn the Vampire output black when focus moves to the other monitor but alas the HDMI switcher doesn’t detect it as loss of sync so does not switch. 

I’m going to try different switchers…

IanSB, 2021-04-25:
@Perifractic 
>Very interesting. How can I test this out? Thanks! (N00b instructions appreciated)

There isn't really any easy way to try it on c0pperdragon's Amiga board. You would have to either use some software to disable syncs which might not be possible at all or put a switch or logic gate in the sync signal path going to the board or the Pi zero to manually switch off the sync signal. I mainly implemented it to confirm that I could put the HDMI output into DPMS standby from bare metal.

It does potentially have some usefulness with the external version of RGBtoHDMI used on such machines as PC CGA / EGA or Amstrad CPC as those use a separate power supply for RGBtoHDMI so with this mod enabled the monitor would go into standby or switch to another input when the computer was powered down even if RGBtoHDMI was still powered.

Perifractic, 2021-04-25:
Thanks anyway @IanSB. What I can say from the Vampire forums is there are a _lot_ of people who'd love to have both devices switch off when not in use. I was surprised. 

In the meantime I'll finish up my video. Do let me know if any ideas come to mind. Thanks everyone!

IanSB, 2021-05-09:
@Perifractic 

I just watched your video. You mentioned to c0pperdragon that the Amiga screen goes grey when not in use so detection of that might be the only way for RGBtoHDMI to go into standby as suggested by LinuxJedi above. Can you post a screencap of that grey screen? (Short press of the button, file will be on the SD card).

LinuxJedi, 2021-05-09:
> I just watched your video. You mentioned to c0pperdragon that the Amiga screen goes grey when not in use so detection of that might be the only way for RGBtoHDMI to go into standby as suggested by [...]

@IanSB I'm also thinking with your DPMS code the CPLD could probably do this and disable sync (with a CPLD version). Saves load on the Pi.

I can also get PiStorms RTG running during the week and will see what colour that gets blanked with if that helps.

Perifractic, 2021-05-09:
@IanSB Screenshot as requested
[image]

IanSB, 2021-05-10:
@Perifractic 
Try this version:
[GreyScreenDPMSBlanking-Beta1.zip](https://github.com/c0pperdragon/Amiga-Digital-Video/files/6448653/GreyScreenDPMSBlanking-Beta1.zip)

It's had limited testing because I don't have a way of setting the grey screen but it might work.
It should put the HDMI output into standby if it sees 200 lines of video set to grey (RGB = 0xAAA based on your screencap) for more than 200 frames ( about 3 to 4 seconds)

@LinuxJedi 
The CPLD is probably too full so some features would have to be removed to fit anything new in.

IanSB, 2021-05-10:
@Perifractic 

If the standby works, you will need a priority HDMI switcher which always switches back to the master (Amiga) input when that is active and only switches away to the other (Vampire) input when the master is in standby.
A lot of HDMI auto switchers will switch to an active input if the current input goes into standby but they might not switch back again unless the second input goes into standby which doesn't happen with the Vampire.

IanSB, 2021-05-10:
@Perifractic 
Does the monitor go into standby when the screen blanks?

This priority switch from Startech seems to fit the requirements and it looks like it is available from Amazon & ebay:
https://www.startech.com/en-us/audio-video-products/vs221hdq
https://www.amazon.com/StarTech-Switch-Automatic-Priority-Switching/dp/B01N6AM2EA
https://www.ebay.com/itm/313515451662

The manual is here:
https://sgcdn.startech.com/005329/media/sets/vs221hdq_manual/vs221hdq.pdf

It states:
`Using the Priority switch feature, you can automatically switch between video sources simply by turning off and on the devices connected to the prioritized ports.  For example: If you typically watch TV through your Cable or Satellite set top box, but sometimes watch movies on your Blu-ray player.  You can connect your set top box to Port #1, and your Blu-ray player to Port#2.  When the Blu-ray player is turned on to watch a movie the HDMI switch will automatically switch to the Blu-ray player.  When the movie is over, simply turn off your Blu-ray player and the HDMI switch will automatically switch back to your set top _box.`

In this case the RGBtoHDMI would be the priority input on Port#2 like the Blu Ray player.

However, I'm not totally sure if the DPMS standby used by the Pi is the on/off state that's being detected by the switch so the only way to confirm that is to try it.

Perifractic, 2021-05-10:
My monitor doesn’t have a standby mode like that so I can’t answer that question however I have ordered that switcher and will let you know as soon as it arrives. Thanks!

LinuxJedi, 2021-05-10:
> It's had limited testing because I don't have a way of setting the grey screen but it might work.
> It should put the HDMI output into standby if it sees 200 lines of video set to grey (RGB = 0xAAA based on your screencap) for more than 200 frames ( about 3 to 4 seconds)

In theory this should work for most RTG solutions too. From what I've been told the Amiga RGB basically switches to the Workbench background colour. For most people that is 0xAAA (there are some installs that use black and it is possible to customise).

> The CPLD is probably too full so some features would have to be removed to fit anything new in.

Fair enough, I don't use the GUI often when compiling Xilinx so didn't really check the usage on it last time I made a JED.

IanSB, 2021-05-14:
@Perifractic
@LinuxJedi  

I just received one of these Startech switchers which I got from ebay. It's a slightly different model (VS222HDQ) which is the dual output version of the VS221HDQ I mentioned above but it is otherwise identical.

It has the same priority switching and it works!!! although I had to connect the two sources the other way around (i.e. Pi zero into input 1 and Vampire into input 2) so try them both ways with your switch (The manual says to use input 2 for the priority same as the switch above but it looks like that is wrong.

Perifractic, 2021-05-14:
Fantastic news! I just got mine and will be trying it today!

Perifractic, 2021-05-15:
@IanSB @c0pperdragon IT WORKS! Under all scenarios. I was impressed that it isn't even necessary to initially select the Vampire when booting into RTG Workbench, or to select the RGB2HDMI when booting games from Gotek. Whether doing that, or running WHDLoaders from WB, as Captain Picard said, "It just works." ;-)

Great teamwork gentlemen. Here is the exact device I bought: https://amzn.to/3ojWscI

Your friend in retro, Perifractic

IanSB, 2021-05-16:
@Perifractic 

Glad to hear it's working OK.
I've incorporated the change into the main software so it will be supported in future releases.

The blanking is always enabled in the test version above, but because it might affect other systems not using auto switching, the functionality will be disabled by default in future releases and you have to enable it by loading the config.txt file from the SD card into a text editor and changing the line:

hdmi_blanking=0
to
hdmi_blanking=1

Then save the file back to the SD card.
Its is implemented this way in the latest main beta available here:
https://github.com/IanSB/RGBtoHDMI/releases
Can you check that works OK for you once you have made the above edit to config.txt
You will have to repeat the process if you ever need to upgrade to a newer release.

Perifractic, 2021-05-16:
Understood. I will check when I can. Good idea to make it optional. Any chance it could be an option in the on screen menu instead of requiring text editor for people? Great work either way!

abrugsch, 2021-05-16:
@IanSB might it be possible to also include a config option to set the colour to switch on? As @LinuxJedi  said it's the background colour which is usually 0xAAA but not always

IanSB, 2021-05-17:
@Perifractic 
>Any chance it could be an option in the on screen menu

That's been implemented and Beta25 can be downloaded here:
https://github.com/IanSB/RGBtoHDMI/releases
Adjust the "HDMI Grey Standby" option in the settings menu to "On" and then save the configuration
Note: If your machine is capable of switching between PAL and NTSC modes you will have to save the setting for each mode as they are separate profiles.

Please give this one a try instead of Beta24 when convenient.

IanSB, 2021-05-17:
@abrugsch 
>might it be possible to also include a config option to set the colour to switch on?

At the moment that is more difficult because there are no spare registers in the capture loops and the AAA value is hard coded as an immediate constant. It might be possible to rework the code sometime but it is not an easy fix.

Perifractic, 2021-05-17:
Will try when I can. Thanks so much!

Perifractic, 2021-05-18:
@IanSB @c0pperdragon BTW here is a short video I'll be making public in the next 2 weeks about the solving of this issue: https://youtu.be/I3_5P_k_6VE

Xnyle, 2022-02-12:
@c0pperdragon I just tested this with a PiSTorm / Emu68 and it always blanks after 3 seconds. Even if I boot a 1.3 Workbench where there is def. no grey anywhere on screen.

LinuxJedi, 2022-02-14:
> @c0pperdragon I just tested this with a PiSTorm / Emu68 and it always blanks after 3 seconds. Even if I boot a 1.3 Workbench where there is def. no grey anywhere on screen.

You may have something else going on here.

> Also I'm not sure if looking for grey is a good idea, early startup menu is also grey.

Early startup menu has content on the screen, every pixel isn't identical, so it will not trigger.

> Why not xor adjacent pixels and as soon as you get a 1 consider screen to be non blank and inc another register. only if that register goes to zero, blank screen. at end of frame dec register. set register with min value from config as soon as above condition is met and you have a configurable timeout This would work for any color and not cost many cpu cycles?

What you are describing isn't too far removed from how it works now. A fixed colour number is needed because a full green screen, for example, is very useful for the user to see.

Xnyle, 2022-02-14:
> Early startup menu has content on the screen, every pixel isn't identical, so it will not trigger.

Ah Ok, than I misunderstood how it's currently working.

> You may have something else going on here.

Must be, but what? It blanks after 3 seconds no matter what if I enable that option.

Additional question: does it only blank the screen or does it also go into a power safe mode (only checking for signal once a second or so). Asking this as I noticed the Pi to be the hottest thing inside the case, even when blanking is active. hotter than the PiStorm pi itself which is kind of strange.

LinuxJedi, 2022-02-14:
> > You may have something else going on here.
> 
> Must be, but what? It blanks after 3 seconds no matter what if I enable that option.

I'm not quite sure. Is RGB output working OK? Does the RGBtoHDMI work with a normal 68K?

> Additional question: does it only blank the screen or does it also go into a power safe mode (only checking for signal once a second or so). Asking this as I noticed the Pi to be the hottest thing [...]

With this enabled it should go into DPMS and it will monitor for changes in the incoming picture. If you are using a software version older than 24th December the Pi is likely to get quite warm.

IanSB, 2022-02-14:
@Xnyle 
Call up the info menu, use the save log & EDID and post log.txt

Xnyle, 2022-02-14:
@IanSB and how do I accomplish that if the screen is off? As soon as I enable grey option screen blanks after 3 seconds.

Edit, Ok, will try to memorize required button clicks.

Regarding questions above, yes also happening without PiStorm, Yes newest FW from 24.12.

Xnyle, 2022-02-14:
https://pastebin.com/JQsqvNeW, Screen turns back on on menu, didn't know that.

IanSB, 2022-02-14:
@Xnyle 
OK it's a bug that only affects the c0pperdragon style interface and it works with the CPLD version.

IanSB, 2022-02-14:
@Xnyle 

Update to the latest beta53 here:
https://github.com/IanSB/RGBtoHDMI/releases

Then overwrite kernelrpi.img  with the version in this zip:
[kernelrpi-blankfix_aplha1.zip](https://github.com/c0pperdragon/Amiga-Digital-Video/files/8064398/kernelrpi-blankfix_aplha1.zip)

IanSB, 2022-02-16:
@Xnyle 
The above log.txt doesn't show any attempt to set the DMPS state so if the screen is going dark that seems to be just the Amiga.
Which model Amiga are you using? Does the analog output go grey or black?

Xnyle, 2022-02-16:
A600 with 3.2 Rom it goes dark grey / blackish but you still see the black borders left and right outside of the 4:3 frame. and then 2 sec later it goes pitchblack but as you wrote no DPMS switch.

IanSB, 2022-02-16:
It's looking for light grey AAA which is this: 

[image]

Try capturing a screen yourself during the 2 seconds (middle button - png file on SD card) and then you can use MS Paint to examine the RGB levels using the colour picker tool and the colour editor.
The value should be 170 (AA) for R,G,B. If you don't get that then it will never trigger.

Maybe different ROM versions behave differently.

Xnyle, 2022-02-16:
There must be another problem:

I just checked, pressing capture every second:
For the first second I'm getting 11 and then AA just like you uploaded it. And it stays like that.

BUT that would mean:
Either it doesn't blank and I should see that picture
Or it does blank and the monitor should turn off

But what is actually happening is what you see in the log:
It doesn't blank
And I see a black screen

I can then capture that screen and later see from the capture that it was actually grey

But the screen i see is pitch black but still on.

So there must be something else going on.

Xnyle, 2022-02-16:
[image]

IanSB, 2022-02-16:
@Xnyle 
>Maybe ther's something going on with sync on SuperDenise?

Looking again at the log.txt, it's changing from 312 line PAL to 525 line interlaced NTSC  
```
Lines per frame = 313, (313.001)  
Autoswitch match: Amiga_50Hz (0) = 63679, 64318, 313, 4  
...  
Lines per frame = 525, (524.999)  
Autoswitch match: Amiga_60Hz_PAL (3) = 63819, 64460, 263, 4  
```

So it's switching to the Amiga_60Hz_PAL sub-profile (525lines but using PAL crystal) which doesn't have the option set.
Try the following:
Change Auto Switch to Off and manually select the Amiga_60Hz_PAL sub-profile.
Change the DPMS setting to On
Save the configuration
Change Auto Switch back to sub-profile only
Save the configuration

Now DPMS will be set on both profiles but it might still not work for other reasons if the signal is unstable (I'm seeing interlace changed a lot in the logs)

Xnyle, 2022-02-16:
Unfortunately no change.

https://pastebin.com/dqR5fnDb

This is the saved 60Hz_PAL Profile

sampling=0,0,0,0,0,0,0,0,0,0,2,0,0,0,0,0,0,5,1,0,55,20,55,20,25,127,127,127
geometry=136,34,640,200,712,224,2,5,1,2,14187580,910,5000,263,4,1,1
hdmi_standby=1
palette=RrGgBb_(EGA)
ffosd_overlay=1

IanSB, 2022-02-16:
@Xnyle 
> Edit: After cold shutdown it's now working. 👍

Shutdown of the converter or the Amiga or both?

Xnyle, 2022-02-16:
Both

aotta, 2022-03-14:
I tested mi A600 vampire with latest driver and 3.1.4 amiga rom, and i get black screen, not gray, when rtg is not used. No way to add a parameter in config.txt?

closed 2022-12-13 by c0pperdragon

mvickers03, 2025-02-13:
Hi, 

I joined just to add my 2 cents as I can rarely help in these situations. 

The colour you see when amiga is outputting to another source can be set using amiga workbench colour pallet.

For example I have a purple theme set up on my Amiga, and when the signal switches to the other screen, I see purple not grey. 

You should be able to change this colour using pallet prefs. This will likely effect some users if you are targeting a specific RGB value.

LinuxJedi, 2025-02-13:
Hi @mvickers03,

Yes, you can find the option to set the RGB value in the menu. The default being the default AAA.