c0pperdragon/Amiga-Digital-Video #50: Flashing of screen from amiga to black when change ntsc amiga 500 to pal 50hz for games

roarfrow, 2021-05-13:
Is there a setting in the menu that is known to fix flashing of screen on amiga 500 when i put my  ntsc amiga 500 in 50 hz pal mode. I didnt want to mess with the settings too much until i asked if this is a known issue with a solution. I will continue to trouble shoot on my side, but any help would be appreciated. its just a stock 500 rev 6 mb 1 meg ram, fat agnus so does pal, but only have 2.10 kickstart. i was able  to rekick into 1.3 but same issue when used degrader to change to pal 50 hz.

roarfrow, 2021-05-13:
just tried it on another tv and same issue. I have another 500 i can test it on, so if nobody has any ideas in the next few days ill see if i get same result on the other machine. thanks for any help!

Bloodmosher, 2021-05-13:
My guess is that either you have auto switch turned off and need to manually switch to the PAL/50hz sub-profile, or you have auto-switch enabled and it is not matching the profile. The code looks for settings that match within tolerance values and I've found that the clock-tolerance range isn't wide enough to auto-switch in the default profile. I run with auto-switch set to "sub-profile" and then in my PAL profile setting I have increased the clock tolerance setting in the "geometry" menu to something larger than the default of 5000. I have it set to 10000 and with this setting I can switch the Amiga between PAL and NTSC and it happily chooses the right sub-profile automatically. You can edit your profile setting file manually to change the setting, e.g. this line:

(for PAL profile):
geometry=136,35,640,256,712,270,1,2,0,2,14187580,908,10000,313,4,1,1

roarfrow, 2021-05-14:
Thank you, that works. I never would have figured it out without your help.

Bloodmosher, 2021-05-14:
Also sounds similar to this thread - https://github.com/hoglet67/RGBtoHDMI/issues/206