c0pperdragon/Amiga-Digital-Video #51: Colors not right

manterola, 2021-05-20:
I am having this problem in a A500+ PAL. The colors are off and I have tried all the options in the settings.
I used the Amiga test kit to help me see where the problem is, and I discovered that the red gradient is not correct.
I have attached a couple of screenshots.
Anyone with similar problems?
Any clue will help me.
Thanks!
[image]
[image]

IanSB, 2021-05-22:
Looks like bit 2 of the red channel is not connected so you need to check the soldering of the surface mount component U3 on pins 4 and 17 which are the input and output of red bit 2

manterola, 2021-05-24:
Thanks...Yes you are right: that red gradient breaks the progression: 0x00, 0x11, 0x22, 0x33, etc. 
After inspection, the problem wasn't located in the daughter board, it was in the RPi0 itself: Pin #32 is grounded.
I removed the pins around the pin #32 and I am still trying to find a bridge. I remember connecting the board on top of the RPi0 in the wrong direction the first time. Do you think that this action affected and burned the GPIO in the pin #32?
Thanks again for this great development

IanSB, 2021-05-24:
@manterola 
>Do you think that this action affected and burned the GPIO in the pin 32?

It's possible but the only way to be sure would be to try another Pi. It might be worth checking the operating temperature of the Pi (System summary in the info menu) because I have a damaged Pi and that runs much hotter than a normal one (~90C vs ~50C)

BTW are you sure you are checking pin 32 because if you count from the wrong end that pin would be a GND.

manterola, 2021-05-28:
Thanks @IanSB . The issue was resolved. I could not find any solder bridging the pin 32 to ground. I desoldered the pins, and followed the traces to see what was wrong. Nothing found, and still pin 32 has connectivity to ground. So I got a new RPi0 today, I checked that pin32 was not grounded and soldered the pin header, and it worked perfectly!.
Thanks for helping me diagnosing this problem.

closed 2021-09-14 by c0pperdragon