c0pperdragon/Amiga-Digital-Video #53: Not possible to get stable image on A500.

PaulKlasmann, 2021-07-02:
I have made 5 amiga 500 adapters. The first two I tried in an A500++ with Super Deniese and correct jumper setting does not result in a stable image on both a  monitor or TV with HDMI.  I see two screens vertically with no sync and two or apprx two screens horizontally on the monitor.  On the TV I see the same non syncing but just two bad images vertically. I'm using v2 software on the RPI zero. 

When I press the switch I get a stable menu. 

I put the adapter onto a DIL socket to protect the pins pn the adapter PCB. Is this a known issue?

PaulKlasmann, 2021-07-04:
Just a thought, how significant is that 3k3 resistor? If it was 10k would it matter? I cant see what its for besides tying GPIO18 to ground. It turned out that 10k Ohm resistors were in the 3k3 container! I can't see that being the cause of the problem.

IanSB, 2021-07-14:
Can you post a screenshot of the source summary page in the info menu

PaulKlasmann, 2021-07-14:
I won't be able to do that now because I removed the flip flop ICs.  It turned out I ordered the part ending in a 3, not a 4.  The correct part has a clock input to clock the outputs so I think the outputs changed as soon as the inputs changed rather than all being clocked at once.  The ones I had had a latch enable input in place of the clock input.  I will update when I receive the new ICs and try the RGB2HDMI PCB again.

PaulKlasmann, 2021-07-19:
I can confirm that the RGB2HDMI is now working correctly.  The D-type flip flops were replaced with the correct clocked version and it works very well.  Sorry about that! Should we delete this issue?

closed 2021-07-19 by PaulKlasmann