c0pperdragon/Amiga-Digital-Video #54: HDMI vs. DVI

bkodenkt, 2021-07-12:
Hey everyone,

I am experiencing a problem with my older adaptor in a A500 rev5 NTSC system. When connecting it with an HDMI cable to my display, everything just works flawlessly and the image is crisp and stable. Due to input limitations, I need to use DVI on that display in the long term, though. With an HDMI to DVI cable on this system, the image is only stable when I open the RGB2HDMI menu. As long as the menu is open, everything is okay, but as soon as I close it, the screen gets black, picture appears again with some wild single vertical line of flickering pixels to the left, screen goes black again, repeat. I checked the cable and hooked up one of the newer CPLD-based adaptors and my A600 PAL system, and this works awesome with both HDMI and DVI inputs. So... it either is a problem with NTSC on my A500 and the DVI or some crazy setting I need to adjust. But... which setting could it be? Any ideas?

Thanks!

bkodenkt, 2021-07-13:
What I tried to fix the problem in the meantime:

- I made the 1mb ChipRAM-hack waaaaay back in the days, but I never adjusted the PAL/NTSC switching on the 8372 Agnus. I just didn't know about this in 199x, and thus my PAL crystal'd A500 always ran in slightly off-timed NTSC modes up until today. So, I decided to put a little piece of tape on pin 41 to constantly have it disconnected so the A500 would be a PAL machine running PAL modes again. While this fixed a problem that drove me nuts back then (Workbench height was 20% smaller due to 200 NTSC-lines instead of 256 PAL-lines), it actually made my DVI-on-off-display-problem worse: The screen now even gets black once every second when the Pi menu is active. When connected via HDMI, everything works fine, though.
- I adjusted the geometry timings as mentioned by @Bloodmosher in this issue #50, and while setting the clock from 5.000 to 10.000 **significantly decreased** the number of display on/black cycles, it made them more unpredictable. The system does run okay for five, ten or 15 seconds, then goes black again, back on, instantly black again, goes back on for like four seconds, then turns black again and again and again and then runs stable for ten seconds... I adjusted from 10.000 up to 15.000 and 20.000, but this made no (visible) difference. Again, when I just use a simple HDMI cable, everything works fine without any flickering.
- I tried @hoglet's latest stable release and @IanSB's latest beta 36, as they both include a 50MHz NTSC sub-profile for my Amiga, but when selecting this, the display just starts flickering a little and it is reverted back to the "normal" 50Mhz profile. Not a surprising outcome, though, as I now have a PAL/PAL system.

In the end it seems there is a problem with my adaptor/DVI combination. On my A600 with a newer CPLD-adaptor and the latest stable release from hoglet, DVI and HDMI work without any issues with the same cables. So, are there known problems with older adaptors from the pre-CPLD era and DVI?

IanSB, 2021-07-14:
Maybe your DVI monitor doesn't support 50Hz display. Hold the button down during power up to force 60Hz output. 
If that works, change the refresh setting in the menu to 60Hz although that will mean stuttery motion on smooth scrolling etc.

Also post a screenshot of the source summary page in the info menu

bkodenkt, 2021-07-14:
> Maybe your DVI monitor doesn't support 50Hz display. Hold the button down during power up to force 60Hz output.
> If that works, change the refresh setting in the menu to 60Hz although that will mean stuttery motion on smooth scrolling etc.

The display does properly support 50Hz on both inputs, DVI and HDMI. I can open and read the 50Hz test message on both inputs. It's the same monitor, just with different inputs. I need HDMI for my Laptop, though... Wait. I could just connect my Laptop via DVI and thus keep the HDMI free for the Amigas instead of the other way round. Heureka!

> Also post a screenshot of the source summary page in the info menu

I will do that, although it seems that I missed the easiest solution for days...

IanSB, 2021-07-14:
@bkodenkt 
>I will do that, although it seems that I missed the easiest solution for days...

Also use the save log function in the info menu after getting your problem display and post log.txt from the SD card