c0pperdragon/Amiga-Digital-Video #56: Atari ST discussion Part 2

c0pperdragon, 2021-09-23:
This is a continuation of https://github.com/c0pperdragon/Amiga-Digital-Video/issues/6

c0pperdragon, 2021-09-23:
@chadders2708

Maybe your previous attempt to sample the both mono pixels using the single extra flip-flop failed because the two sample points were too unmatched. The first points was directly caused by the rising edge of the 16Mhz clock, but the other happend after going through both XORs.

Without properly scoping the incomming signal timings this whole issue can only be guesswork and will be hit-and-miss anyway.

chadders2708, 2021-09-23:
@c0pperdragon 
> Maybe your previous attempt to sample the both mono pixels using the single extra flip-flop failed because the two sample points were too unmatched. The first points was directly caused by the [...]

Yes I came to pretty much the same conclusion. I was planning to scope it out and see exactly what is going on. My old scope died and the new one is still in its box as I'm in the middle of a massive reorganisation / clear out. I'll get things organised and take a proper look at it. 😃

chadders2708, 2021-09-23:
@IanSB 

> That's looking good but one problem is that it won't fully work with the STE models with the 84 pin shifter as they use 12bpp for low and medium res modes: The blanking signal on B3 is not required with those machines as blanking is done digitally in the 84 pin shifter but as all 12 bits are used for the RGB data there is nowhere to feed the mono signal.

It had occured to me that this wasn't going to work for the STE in its current form. I had thought about using the two spare latches on U3 for mono but I was thinking we would need 3 to cover blanking as well. Good news that its done in the shifter on the STE so we don't need a specfic latch for it. I guess we can keep blanking on B3 for the ST and just look at moving mono away from R3 / G3 to allow it to support the STE. 

If I rig this up to feed in mono - say on GPIO 20 / 21? - would you be up for the needed software changes to sample mono from there? 

I can then experiment with feeding the signal directly from the additional latches or via the spare ones on U3 etc. 

Slightly off topic, but while I think about it do you know if simple mode currently has any option to support 3 button inputs? Driving it with a single button is a real pain when experimenting! (its ok for day to day use since its rarely needed). I didn't try but I wasn't sure if the Pi is listening on GPIO19/25 for the other buttons like it does in CPLD mode.

c0pperdragon, 2021-09-23:
@chadders2708
I would be glad to help with theoretical suggestions as I have no Atari myself. It is also fine when you create and sell dedicated kits based on my design.
But I would ask you to refrain from also supporting the Amiga 600 with this. I happen to have about 100 kits left myself for which the market has already pretty much disappeared as it is.

chadders2708, 2021-09-23:
@c0pperdragon 
Many thanks sir. I will naturally credit you accordingly on the silkscreening etc. on any PCB designs. I may produce a handful and sell a few, or use them in the pile of old STs I'm refurbing, but I wasn't thinking of making vast quantities of these or aiming to make any real money from them - I guess I'm more doing this just for the fun of it and the benefit of the user community. 
No issue regaring the Amiga 600 support - can happily leave this out and stick with your existing design there. 😄

IanSB, 2021-09-23:
@chadders2708 
> do you know if simple mode currently has any option to support 3 button inputs?

By default CPLD mode uses 3 buttons and simple mode uses 1 button but if you edit /Profiles/default.txt and change the following setting to:
single_button_mode=1
then the settings get rerversed with CPLD using 1 button and simple mode using 3 buttons.

>If I rig this up to feed in mono - say on GPIO 20 / 21?

I'll have a look at that.

IanSB, 2021-09-25:
@chadders2708 

Try this version:
[AtariST-Alpha3.zip](https://github.com/c0pperdragon/Amiga-Digital-Video/files/7228869/AtariST-Alpha3.zip)

This looks at GPIO22 & 25 instead of R3 & G3 but it is untested.
You should be able to do a basic functionality check before making any mods by pulling either of those two GPIOs high to get alternating black & white vertical stripes like you did with the earlier software. 

WARNING:
After making any hardware changes to use these two GPIOs, don't run older software because those GPIOs will be configured as outputs instead of inputs and you will get two outputs shorted together

GPIO 22 is the analog DAC strobe and GPIO25 drives the mode 7 led in CPLD mode and those two were the least disruptive changes.

chadders2708, 2021-09-25:
@IanSB 

Many Thanks sir! I'll give it a try and let you know.

c0pperdragon, 2021-09-26:
Hi @chadders2708   I just had an idea how to nicely capture the mono:
You could pipe the mono signal through a long series of delay elements (for example with a transparent latch 74LVC573 or something like that). This would give you 9 different slightly phase-shifted variants of the mono.
It should then be fairly easy to find just the right point where to fetch each mono pixel from so their stable times match up with the color sampling.
This would also allow to compensate for variations in the Atari hardware if you provide some means to easily adjust the signal paths during install.

chadders2708, 2021-09-30:
HI @c0pperdragon 
That's not a bad idea at all! Many thanks for the suggestion. I've got a busy couple of weeks now but I will give it a try when I get a bit more time.

Gitbizy, 2021-10-04:
Can someone steer me to the correct gerber files to make a shifter PCB for a standard ST?  Or where to buy a completed board?
What’s the difference between buffered and unbuffered boards for the ST?  Which one is better?

pixelk0, 2021-10-04:
> Can someone steer me to the correct gerber files to make a shifter PCB for a standard ST? Or where to buy a completed board? What’s the difference between buffered and unbuffered boards for the ST? Which one is better?

Given the choice, the buffered board will result in a cleaner, crisper image, removig pixel artifact coming from level instabilities and signal transmission time. I myself only have STes and haven't felt the need for a buffer. The Gerber is here : https://github.com/IanSB/RGBtoHDMI/tree/master/Kicad_Atari_ST_12Bit_Buffered_Pickup/V1
I have 3 unused boards available (because I ordered the wrong buffer chip, and discovered afterward that I didn't need them).

pixelk0, 2021-10-04:
By the way this thread https://www.atari-forum.com/viewtopic.php?f=15&t=40201
contains a lot of info and photos of people like myself who install The 12 bit board in STE, Mega STE and other boards. It may be a valuable asset (maybe some pictures for the wiki ?)

mzryz, 2022-01-20:
@IanSB do you think you might be able to help me with this issue? Thank you: https://github.com/c0pperdragon/Amiga-Digital-Video/issues/62

luciust, 2022-05-22:
@IanSB 
I'm using Beta54: git version: 20220424 34677ba? build version: RGBtoHDMI_20220424_34677 with Atari STF (no-modulator) and Pickup board (no mods). Getting the VSYNC from J13 and HSYNC from C31.

Mono mode (640x400) works just perfect, while colour mode (320x200) doesn't work at all, but... kind-of-works when Amiga 2000 profile is selected. Picture is dimmed and I can see some artifacts:  
[image]

What could be a reason? With Atari ST 50Hz or 50Hz profile I'm getting a black picture...

IanSB, 2022-05-22:
@luciust 
The current profile is set up to assume the blanking mod has been made and without that you will get a blank screen.
(There will be an updated PCB soon with a dedicated blanking input)
Go to the sampling menu and change the sampling mode from "9 bits 0-2 Blanked" to "9 Bits (Bits 0-2)" to use the board without the mod.
(The Amiga profile is not looking for the blanking signal but it is looking at the wrong bits and the wrong sample rate, hence the imperfect image)

luciust, 2022-05-22:
@IanSB Well, I just went on making some changes and - boom - found the sampling solution - wanted to cancel the alarm... But you've already answerd! Wow, this is support - Sunday and quick reply. Thanks, low-res mode works now. Time to get switch to Atari STE
[image]

luciust, 2022-05-24:
With Atari ST (no modulator) and classic IanSB Shifter Pickup board the hi-res mono is "activated" by using i.e. monitor cable that both gives second output.

1. Is the same true for Atari STE with HSYNC and VSYNC used in the same way as with ST?
2. If above is true - has anybody created a "flip switch mod" that would connect monitor port 4 (MONO DETECT) and GND?

IanSB, 2022-05-24:
@luciust 
>Is the same true for Atari STE with HSYNC and VSYNC used in the same way as with ST?

Yes, VSYNC is used for the mono and HSYNC for composite video as all 12 bits are used for RGB on the STE so there were no spare bits for the mono signal on the RGB inputs.
 Grounding the mono detect on the STE should force mono output same as on the ST

>If above is true - has anybody created a "flip switch mod" that would connect monitor port 4 (MONO DETECT) and GND?

Yes, I just bought an ST video connector and glued a small slide switch in the end wired to those pins. Someone else has done that as well and posted a photo but I can't find that post at the moment

aotta, 2022-05-24:
@IanSB i did the video connector with the slide switch hot-glued and posted the photo in one of the ST discussion but can't find it anymore... but here a couple of fresh (and terrible, sorry...) pictures if useful:
[image]

[image]

luciust, 2022-05-25:
Haha, that's cute! But it works!

bwmott, 2023-12-09:
> @luciust The current profile is set up to assume the blanking mod has been made and without that you will get a blank screen. (There will be an updated PCB soon with a dedicated blanking input) Go [...]

@IanSB I've been catching up on this thread again, since I'm finally getting around to doing some more work on my Mega ST. I'm wondering if an updated Atari ST shifter PCB with an input for picking up the blanking signal was created. Looking at the Atari ST shifter PCB in GitHub, it looks like it's Issue 1 from 2021 with pins for H/V signals but not the blanking signal. I looked around, but so far having been able to find a 3-pin version.

M1kerochip, 2023-12-10:
@bwmott, if it's any use, I made a riser, with blanking signal, for the [microATX ST board](http://ataripcb.pl/atari_st_atx_v1.1.html).
[mATX_RGBtoHDMI_Riser REV.1.zip](https://github.com/c0pperdragon/Amiga-Digital-Video/files/13629305/mATX_RGBtoHDMI_Riser.REV.1.zip)

I don't know if that'll fit on a Mega ST. If you've got a pic of the motherboard handy, I can check.

bwmott, 2023-12-11:
@M1kerochip Thanks! Looking at the riser PCB, I think it would fit in the Mega ST motherboard, but would interfere with installing the power supply.

M1kerochip, 2023-12-11:
> @M1kerochip Thanks! Looking at the riser PCB, I think it would fit in the Mega ST motherboard, but would interfere with installing the power supply.

Ah, OK. I see the Mega ST pcb, after work, I'll move the socket to the left side, and move the pins so they don't interfere with the crystal.

M1kerochip, 2023-12-11:
> @M1kerochip Thanks! Looking at the riser PCB, I think it would fit in the Mega ST motherboard, but would interfere with installing the power supply.

Try this?
[mATX_RGBtoHDMI_Riser.zip](https://github.com/c0pperdragon/Amiga-Digital-Video/files/13639112/mATX_RGBtoHDMI_Riser.zip)

I think that should be ok. I added a shrouded idc header, to connect to the rgbtohdmi 12bit board directly.

bwmott, 2023-12-12:
Thanks! I'll order some boards and give it a try.

G4DDS, 2024-01-07:
@aotta just confirming its these two pins here:
[image]

tyristori, 2024-05-17:
I created an adaption of LinuxJedi's A500 CPLD adapter for ST.
[image]
[image]

It gets detected and the CPLD flash screen appears but the CPLD chip is undetected and I can't flash it. Here's my schematic:

[image]