c0pperdragon/Amiga-Digital-Video #60: Output to TV

kevsaenz, 2021-11-14:
I am trying to output the display to a TV 1080 (Sony Bravia 52 inch screen Model:KDL-52XBR) on my monitor the output works perfectly. However, when I hook the system up to the TV I get an onscreen display stating the system is using an "unsupported mode", the Sony site states the TV is capable of the following resolutions 480i, 576i, 480p, 576p, 1080i, 720p, 1080p. Is there a way I can output to a tv made in 2007

c0pperdragon, 2021-11-17:
Maybe your TV does not support a 50Hz image (supposing you have a PAL Amiga).   You can use the settings-menu (using the optional push-button) to force 60Hz output.

IanSB, 2021-11-20:
Try manually selecting the resolution as 1080p and the refresh rate as 50Hz (or 60Hz as appropriate). The default auto setting doesn't produce entirely standard timing and maybe your old TV is a bit more restictive about the sources it supports.
Manually selecting the resolution and refresh will produce exact timings.