c0pperdragon/Amiga-Digital-Video #62: Atari ST Mono LED Flash

mzryz, 2022-01-18:
Hi there,

Sorry if this has been covered before, but I've installed Pi to HDMI in an Atari ST and also an STe but have noticed that the main LED on the board just constantly flashes in mono mode (even though it's done a full calibration) Is there a way I can turn the flashing off?

Thanks,

mzryz, 2022-01-22:
* Crickets * 

It's ok I guess Ians on holiday probably :D

IanSB, 2022-01-22:
@mzryz 
>LED on the board just constantly flashes in mono mode

That's because the mono mode is 70Hz and the hdmi output is 60 Hz so it will never lock

A couple of things to try:
1. Change the refresh to Force "50Hz to Any" which should make the HDMI output 70Hz to match the mono mode. If your monitor can't handle that then you might get a blank screen so will have to hold menu down during reset to recover.
2. Switch off the genlock in the settings menu. This will stop the flashing but the output still won't be locked to the source.

mzryz, 2022-01-25:
Thank you man, much appreciated!

closed 2022-01-25 by mzryz