c0pperdragon/Amiga-Digital-Video #63: Hired Guns - Menu Black Outs

BoozerDawg, 2022-06-16:
I notice with Hired Guns - the opening menus are in a high res interlaced mode, which RGB2HDMI appears to interlaced with Bob De-Interlacing (could be wrong, but its considerably flickery, whereas Agonys intro is flicker free), and it blacks out every couple of seconds, comes back, and then happens again, until you get into the main game. I'm not sure what resolution its actually using here, it could be a mode thats unsupported, just thought I would post to see if there was any information about it :) Thanks!

IanSB, 2022-06-26:
Can you take a screenshot of the source summary page (from the info menu) when displaying that image.
Note the info isn't dynamically updated so you should call up the menu while the problem image is on screen, not before.

BoozerDawg, 2022-06-26:
Yes here you go (I should have done this in the first place), video also - 

[image]

Video showing it - [https://youtu.be/HdPkyMm9XYk](url)

IanSB, 2022-06-26:
@BoozerDawg 

It looks like RGBtoHDMI is detecting that output as progressive sync rather than interlaced sync and it relies on the sync changing from progressive to interlaced to know when to switch on deinterlacing  so it ends up treating the video as progressive, showing each field as a separate frame which results in the shimmer rather than weaving the two fields together.

The "Lines per frame" should show (625 interlaced) if it had detected interlaced sync and you should be able to compare that with some interlaced video that works.

I'm not sure why the screen is blanking but it may be because the sync timing is non standard in some way but it might be possible to get some further info from the log:

After the screen has blanked a few times, use the "Save Log and EDID" option in the info menu and post the resulting Log.txt file from the SD card.

BoozerDawg, 2022-06-26:
Here is the log file - 

[Log.txt](https://github.com/c0pperdragon/Amiga-Digital-Video/files/8987067/Log.txt)

IanSB, 2022-06-27:
@BoozerDawg 

It's seeing some change in the sync which is causing the blanking but I'll have a look into this in more detail when I have the time.

BoozerDawg, 2022-06-28:
Thanks :) It's really not a big deal, its the only game ive seen with this behaviour, but appreciate you looking into it.