c0pperdragon/Amiga-Digital-Video #64: amiga 500 pixelated image

Prince-Eugen, 2022-06-23:
Hi, im using linuxjedi board with cpld for amiga 500, also w rasp pi2 zero.
Settings are default, but picture looks too sharp or pixelated. Is there chance to soften image a bit. It's hurting my eyes on big lcd screen.

here are picture comparison with laggy scart.
[image]
[image]

maybe image need to be interpolated
[image]

Amiga_CPLD_Setup default profile setting
[Default.txt](https://github.com/c0pperdragon/Amiga-Digital-Video/files/8968175/Default.txt)

IanSB, 2022-06-23:
If you have the menu button connected you can call up the menu and change the scaling setting from Auto to Integer(Soft)

abrugsch, 2022-06-23:
In the button menu there are various options for softening and scanline
generation. I find medium scanlines with a little softening looks just
right but your preference might be different