c0pperdragon/Amiga-Digital-Video #65: Unable to activate Single Button Mode for OSD menu

bkodenkt, 2022-06-27:
For some reason, I am unable to switch to single_button_mode. When I short press the button, the menu comes up. When I short press it again, the menu goes away. When I long press it, scanlines are activated. When I long press it again, the scanlines are deactivated.

I found out that I can activate only using a single button by adding single_button_mode=1. To my demise, no guide actually states where to set this. Now, where exactly do I add this config value? In config.txt? configA.txt? default_config.txt? Or inside the profiles folders? And then, in which one exactly?

IanSB, 2022-07-14:
You edit the default.txt file in the /profiles folder on the SD card but if you are using a linuxjedi CPLD board then you should look in the "Amiga_CPLD_Readme" folder on the SD card which contains a readme to setup for correct profile and single button mode.

bkodenkt, 2022-07-15:
I tried a lot of things/files/places to edit single_button_mode=1, but it did not work. As mentioned [here in hoglet67s issue board](https://github.com/hoglet67/RGBtoHDMI/issues/233), simply reverting back to that specific release instantly solved the issue.

maximevince, 2024-04-10:
For anyone still wondering. For some reason `single_button_mode` is inverted for `cpld_version == DESIGN_SIMPLE`, i.e. all the early hardware designs without a CPLD.
See: https://github.com/hoglet67/RGBtoHDMI/blob/master/src/osd.c#L5434

I have no idea what the reason is for this. But setting `single_button_mode=0` effectively _enables_ single button mode on the non-CPLD versions...

It has been introduced in this commit: https://github.com/hoglet67/RGBtoHDMI/commit/7d438d385c86392fa441b5503b9f9ab4e014c66f

Unklemla, 2025-05-06:
Decided to get my Amiga from 1987 updated, so I bought Pi Zero.  How do I get a button added to Pi zero for OSD menu?